esphome/feature-requests #98: Trigger for lights

redskinhu, 2019-03-06:
Hello

**Describe the problem you have/What new integration you would like**
It would be useful if triggers are exist for lights too.

**Please describe your use case for this integration and alternatives you've tried:**
I needed to solve my problem with switches but they are light components anyway and i would like to use them like lights but the HA handled them like switches. Ok i can switch them on/off but the light's features are missing. And i can use HA automations too but what if the HA down....?

**Additional context**
Something like this: ` light.on_turn_on /  light.on_turn_off`

Thx
T

phatpenguintn, 2019-08-01:
A specific use case for this is the Merkury color lights which have to be implemented as a 2 part light since the drivers are different for white and RGB

This is a current working Merkury yaml:
```
light:
  - platform: monochromatic
    name: 'Light1 White'
    output: pwm_w
    id: light1_white
  - platform: fastled_spi
    name: "Light1 Color"
    id: light1_color
    chipset: SM16716
    clock_pin: GPIO4
    data_pin: GPIO14
    power_supply: rgb_power
    num_leds: 8
    rgb_order: RGB
```

it would be great to be able to put something like this on the color light since you need the white light to turn off before you can see the color leds.
```
 on_turn_on:
   then:
     - light.turn_off: light1_white
```

Then you would not have to touch both lights to turn on the color.  

This method would also be great to control the light behavior in different circumstances.
For instance when someone steps onto the porch I want to turn on the white light and then once the pir stops turn off the white and let the color still be on, so I can handle the behavior differently for that light instead of the way I handle the accent lights for example where when white comes on I want to turn off the color.

BTW,
Thanks for all you do!

TimoFriedri, 2019-10-28:
I was really surprised this does not exist yet.

I wanted to use on_turn_on/off to toggle an LED in the power socket to be synchronous to the plugged in light.

*bump*

vinceemail, 2019-11-05:
I would like to request this too. In my use case, I have an RGBW bulb... It is integrated with Home Assistant, and the default action when setting a colour seems to be to turn on the RGB leds, but leave the W led on as well... I think this is an HA issue, but could be solved by automating the RGB and W leds independently, so that when a colour is selected, the W led is turned off. And visa versa.

surpasstop, 2019-11-07:
Same opinion
I want that sonoff light(relay as light) makes turn on/off indicator green light simultaneously

epiniguin, 2020-02-07:
Same opinion: triggers for lights needed 

In the moment I use a workaround:

```light:
  - platform: binary
    name: "cabinet"
    output: relay_1
    id: light_1

switch:
  - platform: template
    name: "Dummy switch"
    internal: true
    lambda: |-
      return (id(light_1).current_values.is_on());
    on_turn_on:
      then:
         ......
    on_turn_off:
      then:
        .......```

closed 2020-04-25 by glmnet