esphome/feature-requests #103: [FR] : SHINYEI PPD42x sensor support

khzd, 2019-03-10:
**Describe the problem you have/What new integration you would like**
I would like to have support for the SHINYEI PPD42x sensor

**Please describe your use case for this integration and alternatives you've tried:**
I implemeted in Arduino platform but woulg like to rationnalise all my old projects in ESPHome

**Additional context**
I'm ready to contribute : coding and testing

khzd, 2019-03-13:
hello
How may i start ?
do you have any defined process ?
cheers

OttoWinter, 2019-03-13:
Have you read the contributing guide?

khzd, 2019-03-13:
no, i didn't but now i did :)

pi4homez, 2019-04-06:
Hi Otto
I'm stick,  I need help on this construction
Have issue with the compilation, As you suggest I restart from the component PMSX003 but stilll have an issue on the definition of Component and sensor , not still clear for me
Thanks

max246, 2023-05-15:
I will tryo to implement this: https://wiki.seeedstudio.com/Grove-Dust_Sensor/#play-with-arduino

karaminder, 2023-11-30:
Any updates @max246? 

I have this sensor inside of my Coway Airmega AP-1512HHS and hoping to intercept the data (and turn the dumb air purifier into a smart air purifier).

Cheers.

jasenpashov, 2024-02-18:
"Hello, is there already support for the 'dust sensor PPD42 / PPD42NJ / PPD42' in ESPhome? Or is there any alternative way to use it? Currently, I'm using them under Arduino."

Cheers.