esphome/feature-requests #108: Remove a device from the frontend (and some other ui improvements)

HA-TB303, 2019-03-12:
**Describe the problem you have/What new integration you would like**

A frontend option to remove a device. Probably best to put it on the bottom of the kebab (3 dots) on the right.

**Please describe your use case for this integration and alternatives you've tried:**

There is no GUI option to remove a device. It would improve user friendliness.

**Additional context**

OttoWinter, 2019-03-12:
See https://github.com/esphome/esphome/pull/438

Until that is ready: The dashboard is just a glorified file viewer. If you remove the YAML file from the config folder, it will not be shown anymore in the interface. Simple as that

OttoWinter, 2019-03-17:
Implemented in https://github.com/esphome/esphome/pull/482

closed 2019-03-17 by OttoWinter