esphome/feature-requests #109: Add support for MLX90614/GY-906 Remote IR Temperature Sensor

inforsgit, 2019-03-13:
**Describe the problem you have/What new integration you would like**

Add support for the MLX90614/GY-906 Remote IR Temperature Sensor. The device is a non-contact infrared thermometer that connects via I2C using address 0x5a and gives two values - the ambient temperature and the temperature derived from the non-contact IR sensor which can be used to measure the temperature of a remote object.

**Please describe your use case for this integration and alternatives you've tried:**

Readout of this sensor is already available in ESPEasy but so far that is the only place I've seen it integrated. There is plenty of information on Sparkfun on how to use the sensor (see below).

**Additional context**

[Sensor description](https://www.melexis.com/en/product/MLX90614/Digital-Plug-Play-Infrared-Thermometer-TO-Can)
[Sparkfun usage guide](https://learn.sparkfun.com/tutorials/mlx90614-ir-thermometer-hookup-guide)

lwqcz, 2019-11-18:
Maybe an Arduino library could help :-) https://www.arduinolibraries.info/libraries/mlx90614
I would appreciate the support of this HW too 👍

lwqcz, 2020-02-19:
@OttoWinter any change to have support for these sensors? Like anytime soon? :-)

tiaanv, 2020-04-15:
it's a year later, but I came across this thread searching for a solution.  Ended up implementing my own.  Please feel free to use it.  It may also be improved upon and integrated into esphome natively if you please. @OttoWinter 

[Github source here](https://github.com/tiaanv/esphome_gy906)

minsuke, 2020-04-18:
> it's a year later, but I came across this thread searching for a solution. Ended up implementing my own. Please feel free to use it. It may also be improved upon and integrated into esphome natively if you please. @OttoWinter
> 
> [Github source here](https://github.com/tiaanv/esphome_gy906)

thx for your great code, but I have i2c gy906 device. can you recommend uart gy906 sensor ?

3gyptian, 2020-12-15:
@minsuke .. in case you're still looking for this I got custom sensor built for gy906 i2c and it's working pretty well ..

https://github.com/3gyptian/esphome-mlx90614-i2c_IR_temp_sensor

roger-, 2021-07-03:
Another custom sensor [here](https://github.com/exxamalte/esphome-customisations/tree/master/mlx90614) (untested).

duojianghuji, 2021-12-10:
> @minsuke .. in case you're still looking for this I got custom sensor built for gy906 i2c and it's working pretty well ..
> 
> https://github.com/3gyptian/esphome-mlx90614-i2c_IR_temp_sensor

It did not work for me. I'm using the ESPHome version(2021.11.4), could u please help?

[image]

tsgoff, 2021-12-10:
> > @minsuke .. in case you're still looking for this I got custom sensor built for gy906 i2c and it's working pretty well ..
> > https://github.com/3gyptian/esphome-mlx90614-i2c_IR_temp_sensor
> 
> It did not work for me. I'm using the ESPHome version(2021.11.4), could u please help?
> 

add the following libraries in this order:
```
  libraries:
    - SPI
    - Wire
    - adafruit/Adafruit BusIO @ 1.9.6
    - "Adafruit SI1145 Library"
    - "Adafruit MLX90614 Library"
```

duojianghuji, 2021-12-10:
> add the following libraries in this order:
>
> ```
>   libraries:
>     - SPI
>     - Wire
>     - adafruit/Adafruit BusIO @ 1.9.6
>     - "Adafruit SI1145 Library"
>     - "Adafruit MLX90614 Library"
> ```

Thanks. Works well

lwqcz, 2022-03-23:
Adafruit library is great source of inspiration, is there any plan to add support for this IR sensor? It would by great to sense PV panel temperature for instance.

closed 2022-07-01 by nagyrobi