esphome/feature-requests #112: Timeout for OTA

macrojames, 2019-03-17:
**Describe the problem you have/What new integration you would like**
Please add a timeout parameter for OTA updates.
I wouldn't care if a terminal or docker process would wait and retry for 30 minutes until the device comes online after the deep sleep.

**Please describe your use case for this integration and alternatives you've tried:**
I'd like to use OTA updates with a deepsleeping node.
The ota_mode flag over mqtt to prevent deepsleep works somehow but has issues:
- you have to wait until the device comes online and is waiting for the update or reset it manually
- you should trigger the update shortly after the device got online because the battery operated node shouldn't be online for a long time
- timing to unset the ota_mode flag on mqtt is crucial, must happen between the device boot and before the device reboot after the update

**Additional context**

```
INFO Successfully compiled program.
INFO Connecting to 192.168.178.225
ERROR Connecting to 192.168.178.225:8266 failed: [Errno 113] No route to host
```
or (depending on host system)
```
INFO Successfully compiled program.
INFO Connecting to 192.168.178.225
ERROR Connecting to 192.168.178.225:8266 failed: timed out
```
This timeframe is about 5- 10 seconds only.

Danke!

OttoWinter, 2019-03-18:
> you have to wait until the device comes online and is waiting for the update or reset it manually

That is how deep sleep works, you can't do anything with it while it is in deep sleep

> timing to unset the ota_mode flag on mqtt is crucial, must happen between the device boot and before the device reboot after the update

why that? if it's a retained message it does not have to happen during that phase.

puuu, 2019-03-19:
Yes, this would be a very nice feature. It needs to be implemented into the esphome cli tool for the `upload` argument. The upload/ota process should be then:

1. publish something ota mode topic (retain)
2. subscribe to device status topic and wait for 'online' (meaning the device waked up)
3. publish empty message to ota mode (to avoid ota mode after device restart)
4. start ota firmware upload

brandond, 2019-03-19:
For use cases like that, I think I'd prefer to have the client pull the firmware down over HTTP. Just have the dashboard server host it, stick the URL in the MQTT firmware update message payload and the client can pull it down when it wakes up.

macrojames, 2019-03-23:
I like the idea to pull a new firmware instead of waiting for a timeframe to push it. Should we open another request for that and close this issue?
Although having a timeout argument wouldn't hurt.

OttoWinter, 2019-03-23:
I don't like the idea of adding another OTA channel, it unnecessarily fragments the OTA system too much. Especially when talking about hosting a HTTP server + then pushing that link via MQTT there are just so many things that need to be set up by the user: Config in esphome, HTTP ports opened, (for example docker), MQTT topic needs to be configured etc.

What I *have* thought about is adding OTA capability to the native API. Especially when native API client mode (then esp is only client, and you have a central api server hosted somewhere; like a mqtt broker but more lightweight and easier to set up). There I think this would be more achievable.

qistoph, 2020-01-31:
> What I _have_ thought about is adding OTA capability to the native API. Especially when native API client mode (then esp is only client, and you have a central api server hosted somewhere; like a mqtt broker but more lightweight and easier to set up). There I think this would be more achievable.

This is something that I happen to have thought of too and tried building. Since then I have found EspHome. My EspSwarmUpdater is just a one man project and I'd rather rely on a tool that has a little more foundation and contributors.

Please have a look at my project:
- https://github.com/qistoph/EspSwarmUpdater/ (Server side component)
- https://github.com/qistoph/EspSwarmUpdater/tree/master/libraries/SwarmUpdater (Arduino library)

I have been focusing on getting it to work with the least amount of configuration and code in Arduino. Ending up with just:
```cpp
#include <SwarmUpdater.h>

SwarmUpdater.update(VERSION_STRING);
```

It uses:
- [mDNS in the client](https://github.com/qistoph/EspSwarmUpdater/blob/master/libraries/SwarmUpdater/src/SwarmUpdater.cpp) to discover the server, then
- `ESPhttpUpdate.update(...)` to check for updates

Could you elaborate on your ideas of the "central api server"? Personally I would prefer the OTA to function with just my local network and without any external dependencies. Did you mean a "central" server in the network, or a single central server somewhere publicly available on the internet?

If you think this is something that could fit ESPHome, I would be enthusiastic to work on implementing it.

jkufner, 2024-09-08:
I think we need to decouple the build and upload steps. When the firmware is built, it needs to be stored until the device becomes online, and then the firmware can be uploaded. We already have the upload mechanism, it just needs to be triggered at a better time.  Then we would not need special OTA modes ot whatever, the OTA component would simply get notified via API when it connects, and then it would do the update as it does now. The difference would be at the end of the build process, where instead of triggering the upload, it would store the firmware somewhere and sent a notification.

schinckel, 2025-03-06:
Since we have the deep sleep data in the configuration, we should be able to set the "retry if unavailable until after X minutes" in the OTA uploader - it compiles the updated firmware, and then attempts to connect to the device. If it fails to connect (and it knows the device is a sleepy one), then it keeps trying until the deep sleep timeout (plus a little bit more) has passed, and then it fails.