esphome/feature-requests #113: Can you add support for VL53L0X distance sensor?

lovinggrey, 2019-03-17:
**Describe the problem you have/What new integration you would like**

**Please describe your use case for this integration and alternatives you've tried:**

**Additional context**

OttoWinter, 2019-03-17:
Please fill out the template

lovinggrey, 2019-03-17:
**Describe the problem you have/What new integration you would like**

可以添加对VL53L0X距离传感器的支持么？
**Please describe your use case for this integration and alternatives you've tried:**

The reason I want to add this sensor is because my washstand needs a distance sensor to determine if someone is standing in front of the washstand and turn on the lighting (my washstand is in the aisle, and often does not need to turn on the light). At present, the distance sensor I use is SR04 ultrasonic distance sensor, but the value of this sensor jumps too much and often turns on my washstand lamp when nobody is there.
**Additional context**

lovinggrey, 2019-03-17:
Sorry, I filled it in. @OttoWinter

vincentkedison, 2019-06-01:
Hi, Any update on this feature request? Thanks

vincentkedison, 2019-06-01:
May this library help, https://github.com/adafruit/Adafruit_VL53L0X

GrowFOUR, 2019-06-04:
I tried using the esphome custom sensor component template but couldn't get it to work

kpenland, 2019-08-05:
I am trying to incorporate this component into my esphome(dev) project.  I have a clean install of ubuntu 18.04 running hassio in docker with the esphome(dev). I understand this PR has not been incorporated in the dev version yet.  I have been successful at compiling a basic firmware -named lir_sensor.yaml and uploading it to my 8266 with LIR , also at including the library VL53L0x . I have put your vl53l0x_sensor files in config/esphome/lir_sensor/src/esphome/components/vl53l0x folder and, while it compiles, I cannot get the vl53l0x in esphome's namespace. 
I need some explicit clubbing with a clue-by-four in order to get this component to work.

redpizza69, 2019-10-12:
Has anyone managed to integrate vl5310x with ESPhome yet? Any time lines? Just got 4 of them and want to setup a trip wire.

kpenland, 2019-10-13:
I have a working sensor using the code in esphome below. I included a custom header file with the custom sensor declaration. 

`substitutions: 
  plug_name: lir_sensor

esphome:
  name: ${plug_name}
  platform: ESP8266
  board: huzzah
  libraries: Adafruit_VL53L0X
  includes: "myVL53L0X.h"

    
wifi:
  networks:
    - ssid: "Housewireless"
      password: "wirelesspwd"
    - ssid: "2nd Wireless"
      password: "7!56%86to5"
    - ssid: "athirdnetwork"
      password: "wireless2"
      
#  manual_ip: 
#    gateway: "192.168.2.1"
#    static_ip: "192.168.2.64"
#    subnet: "255.255.255.0"
# If you're changing ip's on the module and want to upgrade ota to the old ip.    
#  use_address: "192.168.2.64"

  ap: 
    ssid: ${plug_name}
    manual_ip: 
      gateway: "192.168.4.1"
      static_ip: "192.168.4.1"
      subnet: "255.255.255.0"

# Enable logging
logger:
  level: DEBUG
  esp8266_store_log_strings_in_flash: False
# Enable Home Assistant API
api:

ota:

time: 
  - 
    id: homeassistant_time
    platform: homeassistant

i2c:
  sda: 04
  scl: 05
  scan: True
# Example configuration entry
sensor:
  - platform: wifi_signal
    name: "WiFi Signal Sensor"
    update_interval: 60s  
#  - platform: vl53l0x
  - platform: custom
    lambda: |-
      auto my_sensor = new MyVL53L0X();
      App.register_component(my_sensor);
      return {my_sensor};
    sensors:
      name: "LIR Sensor"
# Example configuration entry
text_sensor:
  - platform: wifi_info
    ip_address:
      name: ESP IP Address
    ssid:
      name: ESP Connected SSID
    bssid:
      name: ESP Connected BSSID  

`

Also, I have this code included myVL53L0X.h file,

`#include "Adafruit_VL53L0X.h"
#include "esphome.h"
Adafruit_VL53L0X lox = Adafruit_VL53L0X();

class MyVL53L0X : PollingComponent, public Sensor {
 public:
  MyVL53L0X() : PollingComponent(15000) {}
  void setup() override {
    // This will be called by App.setup()
    Serial.begin(115200);

  // wait until serial port opens for native USB devices
    while (! Serial) {
    delay(1);
    }
  
    Serial.println("Adafruit VL53L0X test");
    if (!lox.begin()) {
      Serial.println(F("Failed to boot VL53L0X"));
    while(1);
  }
  // power 
  Serial.println(F("VL53L0X API Simple Ranging example\n\n")); 
  }
  void update() override {
    // This will be called every "update_interval" milliseconds.
    VL53L0X_RangingMeasurementData_t measure;
    
  Serial.print("Reading a measurement... ");
  lox.rangingTest(&measure, false); // pass in 'true' to get debug data printout!

  if (measure.RangeStatus != 4) {  // phase failures have incorrect data
    Serial.print("Distance (mm): "); Serial.println(measure.RangeMilliMeter);
    publish_state(measure.RangeMilliMeter);
  } else {
    Serial.println(" out of range ");
  }
    
  delay(100);
  }
};`

closed 2019-10-20 by OttoWinter

redpizza69, 2019-10-21:
Thanks kpenland. This worked great.