esphome/feature-requests #114: Mozilla Thing API implementation

blackfyre, 2019-03-18:
**Describe the problem you have/What new integration you would like**

As a user of the ESPHome library, I would like to have the option to use the Mozilla Thing API as an alternative to the Home Assistant API to use custom built devices with the Mozilla GateWay.

**Please describe your use case for this integration and alternatives you've tried:**

Sadly, their Home Assistant alternative only supports that API for communicating with devices. No MQTT, no REST, ... 😢 Or a flexible tool like the ESPHome 😭 

**Additional context**

The people at Mozilla came out with a very versatile & broad spectrum API description which they are aiming to make a W3C standard. It currently exists as a draft at https://iot.mozilla.org/specification/ and their libraries are available for the most popular languages including Arduino C and Python.

OttoWinter, 2019-03-18:
If a home automation system only supports one single API it sounds a lot like it is doomed for failure. I do understand mozilla is trying to make the _api_ it a standard, but the gateway doesn't sound very useful IMO.

blackfyre, 2019-03-18:
While the native support indeed includes only their API, there's a robust adapter framework which enables it to communicate on other protocols & solutions. There are MQTT adapters in progress, but it's been long time coming. Community wise, it's slowly gaining traction as more and more 3rd party adapters emerge.
They are also working on making the Gate Way compatible with a host of other platforms (from docker to a OpenWrt plugin), so most of their effort is going that way.

OttoWinter, 2019-03-18:
Ok, that is good to hear.

From a quick look the base abstractions look a lot like the ones from openHAB - so once openHAB support is coming this could be easier.

davidgraeff, 2019-05-26:
openHAB will change its REST API this year for version 3. (Just fyi, if anyone planned to make ESP Home work via that API.)

I have started to make some noise for Mozilla Web Things (https://community.openhab.org/t/binding-addon-development/75155) to be a second well supported non-OH-specific interface for DIY devices (next to MQTT Homie, https://github.com/esphome/feature-requests/issues/25). Mozillas concept indeed matches openHABs model quite well.

Implementing the Web Things API therefore will allow integration into Mozillas Web Things framework as well as openHAB.