esphome/feature-requests #115: Publish information over the attributes data using native API

tatunts, 2019-03-19:
**Describe the problem you have/What new integration you would like**

Hello! I have a suggestion/question: is it possible to publish attributes data using ESPHome native API? 

**Please describe your use case for this integration and alternatives you've tried:**

This would help a lot to avoid creating multiple entities for sensors related to the main device. For example: if a sonoff basic is configured as a light entity (main relay), wifi RSSI, UPTIME, IP Address, and so on. After connecting the device using the native API, there will many entities created on HA, one for each sensor. This information would be a lot more organized if they could be shown on the main entity, a light entity on this example, as attributes inside the entity card.

Thanks!

**Additional context**

For example (`light.principal_quarto`):

[image]

OttoWinter, 2019-03-19:
Nah sorry this won't be a thing. ESPHome has purposefully stayed away from HA attributes - attributes are quite powerful but that's also their problem. In ESPHome all data has to go through the base types: sensor, binary sensor, light etc. All of them have a single data type (or collection thereof) that they represent. 

Attributes are the complete opposite, the user controls what their types are and what they mean. That gets abused in the HA world much too often. It would also have serious issues for:

 - native API - this expects all data to be well-structured and in defined data types, that cannot exist for attributes
 - config options - every config I can think of would end up being incredibly complicated and not something I want to give to a user.

Attributes are very much a HA thing, and thus should be handled in HA. For example by creating a template sensor that has the state of the esphome sensor and additionally exposes other esphome sensors as attributes.

closed 2019-03-19 by OttoWinter

tatunts, 2019-03-19:
Thanks for you feedback Otto, now I see it better. Have a great day!