esphome/feature-requests #116: turn_on/off_action for switch.gpio

WagnerBeccard, 2019-03-19:
I try for the last days to sync the state of an relais with an led on a tuya plug 
```
switch:

- platform: gpio
  name: "${plug_name}_Relay"
  id: "${plug_name}_Relay"
  pin: GPIO14
  restore_mode: RESTORE_DEFAULT_OFF
  turn_on_action:
  - switch.turn_on: "${plug_name}_LED_Blue"
  turn_off_action:
  - switch.turn_off:"${plug_name}_LED_Blue"

- platform: gpio
  name: "${plug_name}_LED_Blue"
  id: "${plug_name}_LED_Blue"
  pin: GPIO1
  inverted: True
  restore_mode: RESTORE_DEFAULT_OFF
``` 
like its possible with the template switches already.

closed 2019-03-19 by close-issue-app[bot]

WagnerBeccard, 2019-03-19:
nevermind! Finnaly found it! 
on_turn_on - Vote for renaming due consistency. 
I think its very hidden in the wiki.

brandond, 2019-03-20:
@WagnerBeccard https://next.esphome.io/guides/automations.html#all-triggers

OttoWinter, 2019-03-20:
> on_turn_on - Vote for renaming due consistency.

No, **triggers** (like this one) are all called `on_...`. Template **actions** are called `..._action` - that already is consistent.