esphome/feature-requests #120: Command line option to use USB as upload destination

akasma74, 2019-03-20:
**Describe the problem you have/What new integration you would like**
Every time I use
`esphome my_project.yaml run`
or
`esphome my_project.yaml upload`

with a board connected to my PC via a USB programmer, it asks whether to upload a firmware via USB or OTA.
It would be great to have an esphome command line option like --force-usb so it always uses USB to upload.

OttoWinter, 2019-03-20:
That already exists, see cli guide on FAQ please (or -h option)

closed 2019-03-20 by OttoWinter

akasma74, 2019-03-20:
> That already exists, see cli guide on FAQ please (or -h option)

https://esphome.io/guides/faq.html doe not have any information on cli, I was unable to find it anywhere on https://esphome.io

and -h option doesn't help, too:
````
esphome -h
usage: esphome [-h] [-v] [--dashboard]
               configuration
               {config,compile,upload,logs,run,clean-mqtt,wizard,mqtt-fingerprint,version,clean,dashboard}
               ...

positional arguments:
  configuration         Your YAML configuration file.
  {config,compile,upload,logs,run,clean-mqtt,wizard,mqtt-fingerprint,version,clean,dashboard}
                        Commands
    config              Validate the configuration and spit it out.
    compile             Read the configuration and compile a program.
    upload              Validate the configuration and upload the latest
                        binary.
    logs                Validate the configuration and show all MQTT logs.
    run                 Validate the configuration, create a binary, upload
                        it, and start MQTT logs.
    clean-mqtt          Helper to clear an MQTT topic from retain messages.
    wizard              A helpful setup wizard that will guide you through
                        setting up esphome.
    mqtt-fingerprint    Get the SSL fingerprint from a MQTT broker.
    version             Print the esphome version and exit.
    clean               Delete all temporary build files.
    dashboard           Create a simple web server for a dashboard.

optional arguments:
  -h, --help            show this help message and exit
  -v, --verbose         Enable verbose esphome logs.
  --dashboard           Internal flag to set if the command is run from the
                        dashboard.
````

OttoWinter, 2019-03-20:
On the very first entry of the FAQ under point five you have the cli reference: https://esphome.io/guides/faq.html#tips-for-using-esphome

`esphome -h` gives you information about all commands, if you want info about a specific command you need to type `esphome livingroom.yaml run -h` (so the command you would normally type plus -h`, that is standard unix help pages like git does for example.

akasma74, 2019-03-20:
> On the very first entry of the FAQ under point five you have the cli reference

It blends very well within other stuff.. :\

> On the very first entry of the FAQ under point five you have the cli reference: https://esphome.io/guides/faq.html#tips-for-using-esphome
>
> `esphome -h` gives you information about all commands, if [...]

does git require a project name to get help on a specific command?
I expected  `esphome run -h` to do the job, that's what I'd call standard unix help way ;)
Any chance to implement that?

OttoWinter, 2019-03-20:
> It blends very well within other stuff.. :\

True :)

> By the way, my proposal is a bit different - I want to use an auto detected USB port, not a specific upload-port as you suggest.

I don't wnat to add a config option for every single imaginable setup. You'd be better off by using a quick bash script that does `ls /dev/ttyUSB*` and passing that to esphome. Then you can also exclude ones etc.

> Any chance to implement that?

Sam problem exists with version command, but no i don't think it's possible. esphome uses the argparse module and that doesn't have support for that I don't think

akasma74, 2019-03-20:
> You'd be better off by using a quick bash script that does `ls /dev/ttyUSB*` and passing that to esphome

thanks for the tip ;)

> esphome uses the argparse module and that doesn't have support for that I don't think

that's shame. very counter-intuitive. oh well.. :\