esphome/feature-requests #122: Configurable/Substitutions After Flash

dlashua, 2019-03-20:
**Describe the problem you have/What new integration you would like**

I would like to be able to Configure a Device AFTER it's been flashed. This would, of course, be specific to the Device and the Firmware installed. 

* Allow a single Firmware image to be flashed on multiple, identical, devices.
* Allow esphome.name to be suffixed/prefixed with a unique identifier based on MAC
* Allow Wifi AP/Pass, Device Name, and Firmware specific attributes to be configured after flash. 

These should work in the same way that Substitutions work now, however, the configuration data would be saved on the device and reading that configuration would be one of the first things the device would do after boot. 

**Please describe your use case for this integration and alternatives you've tried:**

This would allow one to build/buy 10 identical devices and flash each of them with the same firmware, then (manually at first, with a configuration tool, eventually) configure each of them with their specific properties. Say, I build 10 identical temperature reporting devices. The configuration would include AP SSID, AP Password, Device Name (livingroom_temperature, perhaps, for one of them), and, perhaps, an "update_interval", if the firmware supported it. 

This would allow me to build a firmware, test it, and then release it. As it is now, I can, of course, release a YAML file, but the user would still have to edit it, and compile it before they could flash it.

This would also allow for mass updating of all devices in a network. If the API exposes a firmware type, then a utility could look for all devices with that firmware type and upgrade them with each device retaining its instance specific configuration.

This would also allow a "default minimal firmware" to be created. This would have only a handful of features:

* OTA
* API
* WIFI (with AP fallback)
* An exposed method (probably as part of API, maybe REST) to accept a configuration request, store it on the device, and restart the device.

Additionally, the "ap" mode of any firmware could support these same things. So, for instance, after a boot loop and safemode is engaged, the device would be ready to receive configuration via a smart phone app (for instance) to get it up and running again. 

**Additional context**

I understand that this is not something that can likely leverage the existing Substitutions code. While I'm not well versed in C, I'm fairly certain those substitutions are made before compile. However, the YAML for using "Configuration" variables would look similar to "Substitutions".

As a first (and even as a final result it would be fine) iteration of this feature, the device could refuse to anything more than OTA, Log, Wifi, and API until all configuration variables are set.

YAML might look like:

````
configuration:
  wifi_ssid: string
  wifi_pass: string
  device_name: string
  update_interval: int

wifi:
  ssid: $${wifi_ssid}
  password: $${wifi_pass}

esphome:
    name: $${device_name}
    platform: ESP32
    board: nodemcu-32s

... etc ... etc ...
````

OttoWinter, 2019-03-20:
Duplicate of https://github.com/esphome/esphome/issues/278

No, this won't happen. ESPHome fundamentally depends on knowing everything at compile time. 

For example:
 - How should esphome know where to upload a binary? It doesn't know the hostname so `name.local` won't work.

Also, with this you're essentially disabling all of esphome's validation. ESPHome purposefully validates everything in python, and the C++ core assumes everything has been validated already and no weird values can occur. Your proposal would require duplicating all of the validation process.

Besides, **every single config option** would need to made templatable in the C++ codebase. That obviously would be a monstrous amount of work for a minor annoyance like managing multiple firmware files.

This is how esphome works and what makes it better than alternative products in some areas, and it won't change. 

What *is* possible would be a minimal firmware that cannot be customized that implements only two functions: WiFi and OTA. This firmware can be downloaded from esphome.io and will create a small AP portal where the user can make it OTA-able.

This is already on the list of things to do in the future but a separate issue from yours.

closed 2019-03-20 by OttoWinter

dlashua, 2019-03-20:
Thanks for the response. Based on how ESPHome works, this makes sense. The way it's being presented and used "in the wild" is quite different, which is what led me here in the first place. But I appreciate you helping me to understand the project goals.