esphome/feature-requests #124: Adding LoRa WAN support

stich86, 2019-03-20:
**Describe the problem you have/What new integration you would like**
Add support for LoRa WAN protocol like RFM95

**Please describe your use case for this integration and alternatives you've tried:**

it would be nice to integrate ESP between their self for various automation (like a far remote sensor)

**Additional context**

this is one of the primary library used on Arduino
https://github.com/sandeepmistry/arduino-LoRa

mseefeldt, 2019-05-23:
In the german computer magazine c't was a project that contains an ATtiny and LoRaWan.

Maybe this can be used as an starting point?!
Here is the link to the source code:
[c't Briefkastensensor](https://github.com/jamct/radio-mailbox/tree/master/03_lora_test/lib)

There is even a schematic to attach the LoRaWan module to the ATtiny which can maybe reused for the ESP32.

Lefuneste83, 2019-05-28:
Hi there.

I have been trying to integrate Lora functionalities to Esphome on ESP32 TTGO OLED V1 boards.

I have had some success in sending messages. I am stuck on the receiver part.

Please find bellow my code. Please excuse the limitation of my skills in coding, I am no expert at all in this field.

Sender Side :

lora_sender.yaml file :+1: 

```
esphome:
  name: lora_sender
  platform: ESP32
  board: ttgo-lora32-v1
  includes:
    - lora_sender.h
  libraries:
    - "LoRa"
    - "SPI"

wifi:
  ssid: "myssid"
  password: "mypwd"
  fast_connect: on
  
  manual_ip:
    static_ip: 192.168.0.XX
    gateway: 192.168.0.XX
    subnet: 255.255.255.0

# Enable logging
logger:

# Enable Home Assistant API
api:
  password: 'mypwd'

ota:
  password: 'mypwd'

spi:
  clk_pin: GPIO5
  mosi_pin: GPIO27
  miso_pin: GPIO19

text_sensor:
- platform: custom
  lambda: |-
    auto lora_text_sensor = new lora_sender();
    App.register_component(lora_text_sensor);
    return {lora_text_sensor};

  text_sensors:
    name: "Lora Text Sensor"
```

The custom sensor code is lora_sender.h

```
#include "esphome.h"
#include "LoRa.h"
#include "SPI.h"

using namespace esphome;

class lora_sender : public PollingComponent, public text_sensor::TextSensor {
 public:
  // constructor
  lora_sender() : PollingComponent(15000) {}
  
  void setup() override {
      Serial.begin(115200);
      Serial.println("LoRa Sender");
      LoRa.setPins(18, 14, 26);
      if (!LoRa.begin(868E6)) {
        Serial.println("Starting LoRa failed!");
        while (1);
        delay(1000);
      }
Serial.println("Sending data packet...");
  }

  void update() override {
    // This will be called every "update_interval" milliseconds.
    // Publish state
    publish_state("Hello World");
    LoRa.beginPacket();
    LoRa.print("Hello World");
    LoRa.endPacket(); 
  }
};

```
This is obviously a very simple test case, as it pushes a simple string out. But I believe it is simple enough to adapt it to other cases.

I have validated that the messages gets properly sent by scanning RF with SDR.

Now for the receiving part, I am stuck on the custom text sensor definition. I don't know how to create a sensor that listen continuously (with continuous polling). 

Any help would be appreciated.

OttoWinter, 2019-05-28:
> Now for the receiving part, I am stuck on the custom text sensor definition. I don't know how to create a sensor that listen continuously (with continuous polling).

For that you can override the `loop()` method.

Lefuneste83, 2019-05-28:
Thank you Otto. Sorry for asking but can you point me to some example code ? As I said I am really coming back to coding, as I find your project worth spending some time upon... But I am absolutely a newbie in this field.

OttoWinter, 2019-05-28:
For example `components/gps/gps.h` - It's basically the same thing as [here](https://github.com/sandeepmistry/arduino-LoRa/blob/master/examples/LoRaReceiver/LoRaReceiver.ino) just that the loop function is now part of the component.

Lefuneste83, 2019-05-28:
Thank you Otto I will have a look at this and let you know.

minsuke, 2019-12-13:
> Thank you Otto I will have a look at this and let you know.

thank you for your codes, lefuneste82 and Otto winter's guide. 

Can you share custom codes for the lora_receive.h 

If Lora supports in ESPHOME , It will be more useful platform for the IOT things. 

I wish I can integrate with Lora. 

thank you for your trial Lefuneste83. : )

rradar, 2019-12-25:
I'm also looking forward to get lora supported in esphome. I ordered some **SX1278** modules (Ra-02 Ai-Thinker with 433mhz) to make some tests.

My goal is to connect two different home assistant setup's/networks which are some kilometers apart. Idea is to have a esphome node on each side which are listening to some home assistant (binary)sensors and pushing a message to the other node when a (binary)sensor changes. So it should be possible to mirror states from one home assistant instance/network to another.

rradar, 2019-12-29:
I think I found two very interesting libraries which could be useful for this:

https://github.com/sandeepmistry/arduino-LoRa
**- Support for LoRa Semtech SX1276/77/78/79**

https://github.com/jgromes/RadioLib

Supported modules:
- CC1101 FSK radio module
- ESP8266 WiFi module
- HC05 Bluetooth module
- JDY08 BLE module
- nRF24L01 2.4 GHz module
- RF69 FSK/OOK radio module
- **RFM9x series LoRa modules (RFM95, RM96, RFM97, RFM98)**
- **SX127x series LoRa modules (SX1272, SX1273, SX1276, SX1277, SX1278, SX1279)**
- **SX126x series LoRa modules (SX1261, SX1262, SX1268)**
- SX1231 FSK/OOK radio module
- XBee modules (S2B)

The second one is a bigger (not limited to lora) but very active project

beikeland, 2020-01-09:
> My goal is to connect two different home assistant setup's/networks which are some kilometers apart.

Have you looked at the duty cycle limits for the 433mhz band? You might come close to exceeding that with a somewhat smart smart house.

Johni81, 2020-01-10:
Hey All,
I wanted to share with you my custom code for a lora reciever. I'm using it at the moment as a gateway for two lora nodes, one for my power-meter in the basement, that is based on an arduino pro mini, and the second one as a mailbox sensor, that is based on an attiny84, similar to the one mentioned earlier from c't. I'm using the LoRa library for my RFM95 Lora modules and I included also some AES128 encryption with Crypto/CryptoLW library, the attiny84 limited the encryption to only AES128 due to RAM limits.

lora_gateway.yaml file:
<pre><code>
esphome:
  name: lora_gateway
  platform: ESP32
  board: esp32dev
  includes:
    - lora_gateway.h
  libraries:
    - "LoRa"
    - "SPI"
    - "CryptoLW"
    - "Crypto"
    - "ArduinoJson-esphomelib"

wifi:
  ssid: "myssid"
  password: "mywlanpswd"
  fast_connect: on
  
  manual_ip:
    static_ip: 192.168.178.XXX
    gateway: 192.168.178.1
    subnet: 255.255.255.0

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Fallback Hotspot"
    password: "pswd"

captive_portal:

# Enable logging
logger:
  level: DEBUG

# Enable Home Assistant API
api:
  password: "apipswd"

ota:
  password: "otapswd"
  
# Enable SPI for LoRa
spi:
  clk_pin: GPIO18
  mosi_pin: GPIO23
  miso_pin: GPIO19
  

sensor:
  - platform: custom
    lambda: |-
      auto lora_sensor = new MyLoRaSensors();
      App.register_component(lora_sensor);
      return {lora_sensor->zaehler_sensor_id, 
              lora_sensor->zaehler_sensor_stand,
              lora_sensor->zaehler_sensor_power,
              lora_sensor->zaehler_sensor_elaps,
              lora_sensor->zaehler_sensor_msg,
              lora_sensor->zaehler_sensor_rssi,
              lora_sensor->zaehler_sensor_vbat,
              lora_sensor->mbox_sensor_post,
              lora_sensor->mbox_sensor_msg,
              lora_sensor->mbox_sensor_rssi,
              lora_sensor->mbox_sensor_vbat};
    sensors:
[... 225 lines cut ...]
		receive_buf.replace("\"xxx\"", rssi);
		json_data_pars(receive_buf);
		
		zaehler_sensor_id->publish_state(Z_ID);
		zaehler_sensor_stand->publish_state(Z_Stand*0.0001);
		zaehler_sensor_power->publish_state(Z_Pow);
		zaehler_sensor_elaps->publish_state(Z_Elaps);
		zaehler_sensor_msg->publish_state(Z_msg);
		zaehler_sensor_rssi->publish_state(Z_RSSI);
		zaehler_sensor_vbat->publish_state(Z_VBat*0.000001);
		mbox_sensor_post->publish_state(B_Post);
		mbox_sensor_msg->publish_state(B_msg);
		mbox_sensor_rssi->publish_state(B_RSSI);
		mbox_sensor_vbat->publish_state(B_VBat*0.000001);
		
		for (uint8_t i = 0; i <= MAX_CIPHERTEXT_SIZE; i++)
		{
			old_ciphertext[i] = ciphertext[i];
		}
	}
  }
};
</code></pre>

With this code I'm getting all the sensors publishing the values into HA.

CharlieMaurice, 2020-01-15:
Johni81 do you mind sharing your code for the one of the senders? Im looking to do a mailbox sender with lora, and might take a whack at it with your code. Thanks!

Johni81, 2020-01-15:
@BrewNinja here is my code for the mailbox sender.

<pre><code>
#include &#60;Arduino.h&#62;
#include &#60;ATTinyCore.h&#62;
#include &#60;SPI.h&#62;
#include &#60;LoRa.h&#62;
#include &#60;Crypto.h&#62;
#include &#60;AES.h&#62;
#include &#60;CTR.h&#62;
#include &#60;string.h&#62;
#include &#60;avr/sleep.h&#62;
#include &#60;avr/wdt.h&#62;

// ATMEL ATTINY84 / ARDUINO
//
//                           +-\/-+
//                     VCC  1|    |14  GND
//             (D  0)  PB0  2|    |13  PA0  (D 10)        AREF
//             (D  1)  PB1  3|    |12  PA1  (D  9)
//             (D 11)  PB3  4|    |11  PA2  (D  8)
//  PWM  INT0  (D  2)  PB2  5|    |10  PA3  (D  7)
//  PWM        (D  3)  PA7  6|    |9   PA4  (D  6)
//  PWM        (D  4)  PA6  7|    |8   PA5  (D  5)        PWM
//                           +----+

// these arrays map port names (e.g. port B) to the
// appropriate addresses for various functions (e.g. reading
// and writing)
#define statusled 2    // LED for control
#define irled 1        // power for ir-led of proximity sensor (invisible)
#define irdiode 0      // power in for diode in proximity sensor
#define irsens PA0     //adc pin (connect to output of sensor)
#define csPin 7        // LoRa radio chip select
#define resetPin -1    // LoRa radio reset // -1 for not in use
#define dio0Pin 3      // change for your board; must be a hardware interrupt pin
#define freq 868700000 // LoRa used frequency 868 MHz
#define SF 7           // LoRa used spreadingFactor // ranges from 6-12,default 7 see API docs
#define bw 125E3       // LoRa signal bandwidth in Hz, defaults to 125E3
#define sw 0x12        // LoRa SyncWord ranges from 0-0xFF, default 0x12, see LoRa API docs

// define a threshold for your mailbox.
#define THRESHOLD 15

#define MAX_PLAINTEXT_SIZE 82
#define MAX_CIPHERTEXT_SIZE 82

byte key[16] = {0x01, 0x02, 0x0b, 0x5b, 0xc6, 0x6e, 0xa5, 0xa3, 0xfa, 0x1a, 0xf7, 0xf3, 0x8d, 0xc3, 0x7a, 0xbc}; // The very secret key
byte iv[16] = {0xa7, 0x8a, 0x23, 0x2d, 0xed, 0x1c, 0x77, 0xd8, 0xfd, 0xab, 0x8b, 0x13, 0xc4, 0x8d, 0xb1, 0xf3};
byte ciphertext[MAX_CIPHERTEXT_SIZE] = {0};
uint16_t msgCount = 0; // count of outgoing messages
long vbat = 0;
bool post = 0;
bool post_old = 0;
int sleepCycles = 1;

CTR&#60;AESTiny128&#62; ctr;

/*--------------------------------------------------------------------------------
                                    Setup
--------------------------------------------------------------------------------*/
void setup()
{
  Serial.begin(9600); // TX is AIN0, RX is AIN1
  pinMode(irled, OUTPUT);
  pinMode(irdiode, OUTPUT);
  pinMode(statusled, OUTPUT);

  LoRa.setPins(csPin, resetPin, dio0Pin); // set CS, reset, IRQ pin

[... 166 lines cut ...]
  checkLetter();

  if (post_old != post) // Wenn Wert geändert, dann senden
  {
    getVbat();
    encode_msg(json_data_prep());
    sendMessage(ciphertext);
    post_old = post;
  }

  if (post == true)
  {
    digitalWrite(statusled, HIGH);
    Serial.println("1");
  }
  else
  {
    digitalWrite(statusled, LOW);
    Serial.println("0");
  }

  setSleeptime(900); // Sleeptime in seconds; teilbar durch 8 seconds
  gotoSleep();      // Sleep for x sleepCycles; 1 sleepCycles = 8 seconds
}
</code></pre>

BrewNinja, 2020-01-15:
Appreciate it! Thank you!

rradar, 2020-01-19:
I wonder if the title should be changed to "**Add LoRa support**". Because as far [as I understand](http://www.libelium.com/development/waspmote/documentation/lora-vs-lorawan/) is it all about LoRa here and not about LoRa WAN (yet).

Maybe the LoRa support could be added to the already existing [remote_receiver](https://esphome.io/components/remote_receiver.html) and [remote_transmitter](https://esphome.io/components/remote_transmitter.html) components of esphome?

rradar, 2020-01-31:
@Johni81 thank you very much for your sketch.

I was able to get a wemos d1 mini with a [Ai-Thinker RA-02 433 Lora Module](http://wiki.ai-thinker.com/_media/lora/docs/c048ps01a1_ra-02_product_specification_v1.1.pdf) up and running as a receiver with your code. Little modifications for my 433Mhz module in lora_gateway.h are:

```
#define csPin 15       // LoRa radio chip select
#define resetPin 16    // LoRa radio reset // -1 for not in use
#define dio0Pin 5      // change for your board; must be a hardware interrupt pin
#define freq 433000000 // LoRa used frequency 433 MHz
#define SF 7           // LoRa used spreadingFactor // ranges from 6-12,default 7 see API docs
#define bw 125E3       // LoRa signal bandwidth in Hz, defaults to 125E3
#define sw 0x12        // LoRa SyncWord ranges from 0-0xFF, default 0x12, see LoRa API docs		!!! MUST be equal on Lora sensor nodes and gateway
```

and my spi pins on the wemos d1 mini are:

```
spi:
  clk_pin: D5
  mosi_pin: D7
  miso_pin: D6
```

My wiring:
```
   ESP8266      -  RA-02
	Gpio15/D8   - NSS
	Gpio13/D7   - MOSI
	Gpio12/D6   - MISO
	Gpio14/D5   - SCK
	Gpio05/D1   - DIO0
	Gpio16/D0   - RST
	VCC         - 3.3V
	GND         - GND
```

Now I wonder what's needed to get another pair of this (esp&ra-02) acting as a sender. My idea is to forward (binary)sensor data the esphome node is knowing (from local or home assistant) anyways.

Could this be possible? :thinking:

TessyPowder, 2020-06-24:
Any Chance this will get properly implemented in Esphome ?

rradar, 2020-06-29:
> Any Chance this will get properly implemented in Esphome ?

I guess the best would to support [RadioLib](https://github.com/jgromes/RadioLib) but I could imaging that this is a bunch of work

philletourneau, 2020-07-11:
+1 desire for an official LORA integration with ESPHOME!

acelauwc, 2020-09-05:
+1 desire for an official LORA integration with ESPHOME!

maxparrela, 2020-10-12:
+1 desire for an official LORA integration with ESPHOME!

cristianm, 2020-10-24:
+1 desire for an official LORA integration with ESPHOME!

Koma-Andrea, 2020-11-10:
+1 
Also if some esphome developer needs a LORA device I'll chip in !

AquaMCU, 2020-11-15:
Hi. I really like to have LORA for my home automation system. It makes my WIFI issues disappear in a second.
Please add the support.
Thank you in advance!

AquaMCU, 2020-11-17:
I have researched LORA a little bit. I think for starters, it would be nice to be able to initialize a LORA module as a sender and receiver.
It would be nice to have something like LORA.send (STRING) and some kind of event when the receiving end gets a message.

That way you could set up a very simple LORA connection between two devices.

Do you think this is possible?

gretel, 2020-11-17:
@AquaMCU no, because this is a feature request.

AquaMCU, 2020-11-18:
I know. So where would you go and post a feature request like this? How can the community fund this development? I think there are some guys interested in this. Do you want to do some kind of  bug bounty or a kick starter?

Oliver

P.s. I already donated 50$ for this to Otto and I would add another 50 to some kind of bug bounty.

BrewNinja, 2020-11-18:
Best bet would be to drop into discord and possibly reddit and see if anyone is interested. As mentioned, radiolib looks like a promising way to integrate, but it would still take someone's time to actually add it. I think this is a no-brainer for adding functionality, but the issue is finding someone who is interested in actually doing the adding.

martin3000, 2020-12-07:
With this library, it is easy: https://github.com/sandeepmistry/arduino-LoRa

To send:
```
LoRa.setPins(SS, RST, DIO0);
LoRa.begin(BAND);
LoRa.beginPacket();
LoRa.print("hello ");
LoRa.endPacket();
```

For receiving:
```
LoRa.setPins(SS, RST, DIO0);
LoRa.begin(BAND);
LoRa.parsePacket();
while (LoRa.available()) {
      LoRaData += (char)LoRa.read();
}

```

atxbyea, 2021-01-04:
@Johni81 Thanks for the examples!

I'm trying to get that to work with a current version of esphome, however by default it installs ArduinoJson6 (which I have worked around by converting the buffer to document per 6 documentation, that seems to work.

After that it seems to struggle wiith the definitions here :+1: 
src/lora_gateway.h:143:3: error: 'Sensor' does not name a type
   Sensor *zaehler_sensor_stand = new Sensor();

And fails compiling.

Is anyone else using this actively ? I have two LoRA modules and have got them up and running with working LCD displays, but trying to add the LoRA side of it with  esphome might not be as straightforward as I hoped.

atxbyea, 2021-01-04:
or by chance do you @Lefuneste83 have the code for your reciever ? If you ever got it running, your sender code appears to work nicely atleast, and compiles as expected

Lefuneste83, 2021-01-04:
Sorry I haven't been working on this matter lately. I moved to OpenMQTTGateway to do some initial connectivity tests as it kinda supports Lora out of the box. But I haven't been much further than having a Lora sender to send a message to the receiver. At least I was able to have it to work. I don't know why it takes so long for ESPHome to fully support Lora protocol as it is really necessary for some use cases. I am interested to setup a mailbox detection system and only Lora would allow such an application without having to imagine crazy systems for power supplies. Other RF systems wouldn't transmit the distance I have between my batcave and the mailbox...

atxbyea, 2021-01-04:
@Lefuneste83 nice! I didn't even thing about OpenMQTTGateway.. I tried to get that running with 433 but gave up, I'll give it a shot!

Johni81, 2021-01-04:
@atxbyea not sure how you have modified the code for ArduinoJson6, I had also to do it sometime ago and my code is still compiling and running. I just checked it once again and it compiling without errors.

Here is what I have done.
<pre><code>
void json_data_pars(String json)
{
  const size_t capacity = JSON_OBJECT_SIZE(8) + 60;
  DynamicJsonDocument  doc(capacity);

  //JsonObject& doc = jsonBuffer.parseObject(json);
  deserializeJson(doc, json);

  String Sens = doc["Sensor"]; // "Zaehler"
  
  if (Sens == "Zaehler")
  {
   ...
</code></pre>

I hope that helps you to solve it.

rradar, 2021-03-29:
Please test this PR: https://github.com/esphome/esphome/pull/1609

quentindavid, 2021-04-07:
Nice feature !! The must would be to include AES encryption and anti-replay counter as well (can be useful if you want to use a lora esphome gateway to open your garage door for example !)
I will give a try with this PR in the meantime !

theuzzy, 2021-07-06:
Hi is there any progress on this?
it would be really awesome to get this working.
If needed testing, I could try and help.

Regards,

LordTompa, 2021-08-06:
@Johni81 could you also upload your code for the electricity meter, i want to use lora for exactly that and i'm at a loss right now

ybarigou, 2022-01-12:
> Please test this PR: [esphome/esphome#1609](https://github.com/esphome/esphome/pull/1609)

Has anyone tried this one with ESP8266 as both sender and receiver ?

dissidence, 2022-04-03:
I would love to see some movement on this, I do support nebucasa if they need more money for this I would be willing to help out more.

pauldogg, 2022-04-03:
> I would love to see some movement on this, I do support nebucasa if they need more money for this I would be willing to help out more.

Same here!

prghix, 2022-04-04:
The absence of LORA support in ESPhome is our handicap.

I'd like to send some $$$ for this issue as well.

CFenner, 2022-04-27:
Is there any work ongoing where it's possible to contribute? Is https://github.com/esphome/esphome/pull/1609 worth looking into to continue the work done there?

gatonero, 2022-05-27:
This topic is getting more and more interesting, because the existing protocols are to delimiting in the sense of range. I'm living in an apartment building and even can't reach my car or cellar.

AquaMCU, 2022-06-03:
Perhaps we need to organize a hackerthon to develop this. I could be there and test on two boards (TTGO and HELTEC).

mmichon, 2022-08-16:
My application is for a LoRa-based temperature sensor (sender) and display (receiver). I have a [TTGO Lora32 OLED](https://randomnerdtutorials.com/ttgo-lora32-sx1276-arduino-ide/) (SX1276) and tried https://github.com/esphome/esphome/pull/1609 with these versions of that PR's [test configs](https://github.com/esphome/esphome/pull/1609#issue-830384125)
```yaml
substitutions:
  esp_hostname: esplora-recv

esphome:
  name: ${esp_hostname}
  platform: ESP32
  board: ttgo-lora32-v1

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

api:
captive_portal:
logger:
ota:
web_server:

lora:
  - platform: sx127x
    id: lora_12
    sync_word: 0x34
    di0_pin: GPIO26
    rst_pin: GPIO14
    cs_pin: GPIO18
    band: 868MHZ

sensor:
  - platform: wifi_signal
    name: "Lora 2 WiFi Signal Sensor"
    update_interval: 5s
  - platform: template
    name: "Lora 2 Showing Lora1 WiFi Signal Sensor"
    receive_from_lora: true
    lora_name: lora1_wifi_signal_sensor
    update_interval: never
    unit_of_measurement: "dB"
    icon: "mdi:wifi"
  - platform: lora
    rssi:
      name: "Lora2 Last RSSI"
      update_interval: 5s
    snr:
      name: "Lora2 Last SNR"
      update_interval: 5s

switch:
  - platform: template
    name: "Lora 2 Showing Lora1 Switch Binary Sensor 1"
    receive_from_lora: true
    lora_name: switch_binary_sensor_1

binary_sensor:
  - platform: template
    name: "Lora 2 Showing Lora 1 Binary Sensor"
    lora_name: lora_1_binary_sensor
    receive_from_lora: true

text_sensor:
  - platform: template
    name: "Lora 2 Showing Lora 1 Wifi SSID"
    lora_name: lora_1_wifi_ssid
    receive_from_lora: true
```

```yaml
substitutions:
  esp_hostname: esplora-send

esphome:
  name: ${esp_hostname}
  platform: ESP32
  board: ttgo-lora32-v1

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

api:
captive_portal:
logger:
ota:
web_server:

sensor:
  - platform: wifi_signal
    name: "Lora1 WiFi Signal Sensor"
    lora_name: lora1_wifi_signal_sensor
    update_interval: 5s
    send_to_lora: true
  - platform: lora
    rssi:
      name: "Lora2 Last RSSI"
      update_interval: 5s
    snr:
      name: "Lora2 Last SNR"
      update_interval: 5s

switch:
  - platform: gpio
    name: "Switch Binary Sensor 1"
    pin: GPIO25
    on_turn_on:
      - binary_sensor.template.publish:
          id: bin1
          state: ON
    on_turn_off:
      - binary_sensor.template.publish:
          id: bin1
          state: OFF
    send_to_lora: true

binary_sensor:
  - platform: template
    name: "Lora 1 Binary Sensor"
    id: bin1

text_sensor:
  - platform: template
    name: "Lora 1 Wifi SSID"
    id: text_info_ssid
    send_to_lora: true
    lambda: |-
      if ( id(my_global_int) == 0 ) {
        id(my_global_int)=1;
        return {"Hello World 1"};
      }else{
        id(my_global_int)=0;
        return {"Hello World 0"};
      }

    update_interval: 2s
```

but got these config errors
```
mmichon@Mikes-MBA ~/s/147-esphome (master) [1]> esphome esplora-send.yml config
INFO Reading configuration esplora-send.yml...
Failed config

sensor.lora: [source esplora-send.yml:25]
  
  Component sensor.lora requires component lora.
  platform: lora
  rssi: 
    name: Lora2 Last RSSI
    update_interval: 5s
  snr: 
    name: Lora2 Last SNR
    update_interval: 5s
mmichon@Mikes-MBA ~/s/147-esphome (master) [1]> esphome esplora-recv.yml config
INFO Reading configuration esplora-recv.yml...
Failed config

lora.sx127x: [source esplora-recv.yml:20]
  
  Component lora.sx127x requires component spi.
  platform: sx127x
  id: lora_12
  sync_word: 52
  di0_pin: GPIO26
  rst_pin: GPIO14
  cs_pin: GPIO18
  band: 868MHZ
```

Feels like I'm close (on this particular board) but something's up with my esphome build. AFAIK the esphome docs don't advise how to build from source and I'm no Python expert. If I can't get esphome to work, I'm just going to build my project with Arduino libs in C++. Bummer!

Baccega, 2022-08-24:
@mmichon Looking at the error message and code you are probably missing the lora component in esplora-send.yml:
```
lora:
  - platform: sx127x
    id: lora_12
    sync_word: 0x34
    di0_pin: GPIO26
    rst_pin: GPIO14
    cs_pin: GPIO18
    band: 868MHZ
```
(copied from esplora-recv.yml)

And the spi component in both the files:
```
# Enable SPI for LoRa
spi:
  clk_pin: INSERT_PIN_BASED_ON_BOARD
  mosi_pin: INSERT_PIN_BASED_ON_BOARD
  miso_pin: INSERT_PIN_BASED_ON_BOARD
```
(You can look at [image] example by   @rradar )

MarkoPaasila, 2022-08-26:
There's also [RadioHead](https://www.airspayce.com/mikem/arduino/RadioHead/index.html)

MarkoPaasila, 2022-08-27:
What would it require for the HA <-> esphome API to work over LoRa instead of Wifi? So that sensor-, button-, switch-, etc components would automatically register in home-assistant, and the esp device would get time and other messages from home-assistant.

vincegre, 2022-08-27:
> What would it require for the HA <-> esphome API to work over LoRa instead of Wifi? So that sensor-, button-, switch-, etc components would automatically register in home-assistant, and the esp device would get time and other messages from home-assistant.

Please inform yourself about what is Lora and how it works before posting such non sense things :-D Lora is in no way similar with Wifi !!

MarkoPaasila, 2022-08-27:
I'm doing my best. And of course it's different from wifi - otherwise we wouldn't need to even talk about LoRa.

MarkoPaasila, 2022-08-27:
I'm doing my best. That's why I ask questions. I already think I understand that it's a wireless transmission protocol for longer range and slower data rates than wifi - which is why it's interesting. By asking I assumed that the esphome devices communicate with home-assistant. Currently it is user-friendly because of the eshome api being integrated in home-assistant. Am I right? But the range is currently limited by the choice of wifi/ethernet. So is it in principle (not currently) possible to have such a nice integration between the two, but over LoRa? Or over whatever other transmission protocol/medium.

I know very little, but I do my best to ask sensible questions. So why didn't my question make sense? Is it not possible for a computer (home-assistant) to communicate with a remote device (esp32) over various means of communication (wifi, ethernet, ble, LoRa, rf868)?

vincegre, 2022-08-27:
> I'm doing my best. And of course it's different from wifi - otherwise we wouldn't need to even talk about LoRa.

that's why you can't use Lora for ESP <-> HA communication ! only for communication between ESP and some sensors and other devices ! Please read basic documentation on how Lora works and you'll understand quickly your mistake ;-)

MarkoPaasila, 2022-08-27:
> only for communication between ESP and some sensors
So LoRa is an alternative to i2c? I thought it was for long-distance low-power communication from a device to another (mcu, computer). Isn't all communication one way? If I communicate to you, it's one way. Then you communicate back. Wifi works like that too. The AP sends a message into the air, and a STA receives it. Then the opposite happens.

vincegre, 2022-08-27:
> So LoRa is an alternative to i2c? I thought it was for long-distance low-power communication from a device(often **with** a sensor) to another (mcu, computer).

Absolutely not ! I2C is a printed circuit board serial communication protocol ! Lora is a very low power very low bandwith system to collect payloads emitted by sensors (can be anything) and as it uses extremely low power it can work during years on a single battery. Lonh distance of it allows to use it without needs of a gateway every 50m like Wifi ! The counterpart of that is that it's 99% transmission from sensors to gateways and in fact lot of Lora devices ary only able to send datas not receive some !

 
> Mainly but not exclusively. It works equally well both ways, with the limitation that the remote device might not be listening 100% of the time. The _mainly_ -part stems from the legal limitation that a device can transmit only 1% of the time and that there are usually more end-devices than gateways - not from technical reasons. And isn't all communication one way? If I communicate to you, it's one way. Then you communicate back. Wifi works like that too. The AP sends a message into the air, and a STA receives it. Then the opposite happens.

You really need to read some good documentations about Lora. It doesn't work equally well both ways as it's designed to handle mainly communication from remote devices to gateways not the reverse ! (here my Lora gw have a ratio of 90% percent traffic from devices to gateways and 10% and most of time more 5% of communication from gw to devices !
You forget also something very important: Lora can be compared to UDP, device sends a message in the air and good luck ;) In Wifi you have a permanent radio communication established in full duplex mode between devices and Access Point/gateways !

Perhaps we should stop there our pollution in this thread ;-)

AquaMCU, 2022-10-20:
Hello y'all.

There was a very promising commit here:

https://github.com/esphome/esphome/pull/1609#issue-830384125

But the author ran out of time. what a shame. Can we not pick up, where he left of and finalize this? Looks quite nice.

Since I am not a c++ guy, I can help with testing. I have two of the boards here and I have home assistant up and running. 

Who can review this and comment, what is missing for integration into esphome.

Oliver

alf-scotland, 2022-10-30:
@AquaMCU I just created a new draft PR based on the original one from SenexCrenshaw. I am planning to put some work into it in the coming days as I am dependent on getting LoRa to work for a project of mine. Maybe the seasoned ESPHome experts here can help provide conceptual input.

I had previously worked on my end with a custom `PollingComponent` and the LoRa package (sandeepmistry/LoRa) to implement LoRa communication between two TTGO LORAv2 ESP32s. This was very "hand-woven" and felt like bending ESPHome into something that it could do better.

I read some comments above about LoRa vs LoRa WAN - and I think the latter is definitively desirable for the community but also a stretch for a single contributor (https://lora-alliance.org/resource_hub/lorawan-104-specification-package/).

So a shout out here I am super appreciative for any design/code input on the just opened draft PR esphome/esphome#3970 🙏

matthew73210, 2022-11-04:
> @AquaMCU I just created a new draft PR based on the original one from SenexCrenshaw. I am planning to put some work into it in the coming days as I am dependent on getting LoRa to work for a project [...]

Hey,

Looking to have Lora working on some node too. For some reason I loose wifi on one or 2 devices depending on the weather. Can't do much as they live outside and and i'd rather not pay for a new AP. What I would like is LoRa to be a backup for only the data. Anyhow if you need some help testing I'll be happy to lend a hand. I currently have 2 TTGO lora V1.3 on hand, one is running openmqttgate but stability is an issue as it dissconnects from wifi. Strangly when flashed with esphome it never goes offline. I digress. I'll have a look at what you're doing and see if I can get it working on my end.

AquaMCU, 2022-11-21:
Hello Folks.

Here is a new proposal. I am not giving up!

---
LORA to MQTT gateway component:
Receiving sensor data:
The sensor sends a string via LORA like "/sensorID/topic/value". This gets parsed in the gateway into topic and payload and is send to the MQTT server using the ESPHomes MQTT component

Sending Data:
The component gets a list of topics and will notify for these on the MQTT server. When receiving a message, it will send the data on lora like above  "/sensorID/action/value"
---

For the nodes, we need a simple component that registers for a list of "/sensorID/action" that can be used with on_xxx statements

As for sending, an action can be called to send the data via lora to the gateway.

What do you guys think?

prghix, 2022-11-22:
> What do you guys think?

I'd vote for anything that would work.

Missing LORA support is IMHO a growing handicap for ESPhome. Wifi or wire is not always available... so it limits the range of applications considerably.

[23 more replies]