esphome/feature-requests #126: Wake On Lan Switch send WOL Packet

kaiomatico, 2019-03-21:
**Describe the problem you have/What new integration you would like**

I would like to be able to define a "wake on lan" switch where I can put in a mac address

**Please describe your use case for this integration and alternatives you've tried:**

My goal is to use esphome with it's web server as standalone smart home central because it includes everything I need (really just temperature other than wake on lan)

**Additional context**

OttoWinter, 2019-03-22:
> My goal is to use esphome with it's web server as standalone smart home central because it includes everything I need (really just temperature other than wake on lan)

While it is possible to use esphome that way, it is in no way supposed to be used without a central home automation system. You _can_ use it this way, it's just not made for this use.

Flagging this a very low priority as any home automation system can do this - and esphome isn't trying to (nor can it ever) replicate the functionality of a home automation hub.

vevsvevs, 2019-08-15:
@OttoWinter, first of all - thank you for such an amazing project!

My HA instance hosted as VM on the home server which power is backed up with APC SURT. Power shortages happen quite often in my apartment, so i often facing such situation: server goes down when UPS battery drops to 5% of charge, and there is no way to power it on automatically because UPS doesn't toggle the load when power is restored. So my entire smart home is dead until i come and power it on manually :( WOL-packet right after connecting to the Wi-Fi sent by of one my ESPHome-based device would be a great solution, please consider to implement this feature!

glmnet, 2019-08-15:
Do you need to WOL your APC? or your PC?
If it is a PC for this scenario I would use a BIOS setting to restore power state after AC is back.

https://lmgtfy.com/?q=bios+turn+on+computer+on+power+restore

vevsvevs, 2019-08-15:
WOL PC of course :) UPS keeps the load turned ON whilst UPS is alive (5% enough for ~10 hours of activity) and does NOT toggle the load after AC is back. So, "restore power state" option is useless, unfortunately, because there is no "power loss" event occurred.

glmnet, 2019-08-15:
Ahh so your PC is turned off by the UPS software.

In that case your last option left is ALWAYS_ON, that is, in any case power is restored the PC will go power itself on.
All BIOS I've seen has the tree options: 
 - Always off, default
 - Recover last state (not applies here)
 - Always on

vevsvevs, 2019-08-15:
Power halting for Windows server with hosted VM's within (including HA and OpenWRT) with Storage Spaces mirror is definitely can't be called "reliable" process, not an option for me :))) And yes, timer-based power presence checking also would be nice (an adjustable pre-WOL timer, f.e.). Hopefully, @OttoWinter will rise up the priority of this issue ;)

glmnet, 2019-08-15:
If you configure it with always on you can still turn your pc off via software from a ups event. 
Then the computer will power on after an ac cycle from ac disconnected then connected. 
It will fail, indeed if power comes back after pc is signaled to turn off and power comes back before the ups switches the ac off.

vevsvevs, 2019-08-15:
The power cycle doesn't occur when the UPS is completely discharged until AC is restored - the load remains off. And this also doesn't happen when the UPS has survived on batteries - the load remains on.

sgvj, 2019-09-13:
I would very much like this too.
Here is my usecase:
I have a computer that turns off over the weekend. When I return to it I'd like the binary BLE sensor to find my bluetooth fob and turn it back on. Let's save electricity!

pullaf, 2019-12-24:
I would also like to request this feature
I recently purchased a "dumb" UPS that doesn't tell me how much charge is left, so I turn off Homeassistant soon after a blackout by monitoring a couple of ESPHome
When the power is back on, and the ESPHome devices are back online, it would be great if they (well, one of them) could try waking up the HA server

RomanWojnar, 2020-01-19:
There is at least a small possibility that WOL will ever be added to ESPhome.
I would also very much welcome this feature.

brgaulin, 2020-04-11:
I'm looking at this as well, might send in a PR. 

Use case:
1. Power goes out
1. Servers gracefully shut down
1. Power comes back
1. ESPHome device somewhere in the house waits 5min after startup(ensure stability), then sends WOL
1. Server turns back on

RomanWojnar, 2021-06-29:
Again over time I want to ask if it is possible to implement the wake on lan function in ESPHome.
Thank you.

xiaolei0125, 2021-07-12:
I would also like to request this feature

koeberlue, 2021-08-01:
I would also love to see this feature:

My usecase

RomanWojnar, 2022-02-15:
Eureka!
WoL works perfectly!
Thank you! Thank you! Thank you!