esphome/feature-requests #127: Individual speed setting for stepper

noxhirsch, 2019-03-21:
**Describe the problem you have/What new integration you would like**
The stepper function only allows to set a max speed in the initial config. It would be great, if you can set an additional speed setting with set_target:
```
on_...:
  then:
  - stepper.set_target:
      id: my_stepper
      target: 250
      max_speed: 250 steps/s
```

**Please describe your use case for this integration and alternatives you've tried:**
I've built a roller blind with a stepper motor. To make it more practical I'd like to speed up the homing sequence and also integrate a slow mode as "alarm" (slowly and silently move up the roller blinds in the morning). At the moment I can only set one "normal speed".
As an alternative I tried moving the stepper a little bit, wait some milliseconds and move further - but that get's very noisy.

**Additional context**

OttoWinter, 2019-03-24:
Hey @claprepi, @rexyiy please use the reactions feature in github instead of writing `+1` (notification spam), I will hide all +1 comments going forward.

OttoWinter, 2019-04-25:
A stepper.set_speed has been added in dev.

closed 2019-04-25 by OttoWinter