esphome/feature-requests #129: Possibility to have LOCK entity

Klagio, 2019-03-21:
I managed to have two switch: one goes ON for LOCK and another switch goes ON for UNLOCK,

but now in frontend I have two switches. I wish I had a single entity "lock"

**Describe the problem you have/What new integration you would like**

**Please describe your use case for this integration and alternatives you've tried:**

**Additional context**

OttoWinter, 2019-03-22:
From what I see, locks are just switches - with the exception that they have both lock/unlock actions visible.

If that is the case, please use the template switch and see `assumed_state`.

If locks are different in another way (that is not just visual, that should be done in HA), then reopen this issue

closed 2019-03-22 by OttoWinter

brandond, 2019-03-22:
It kinda sounds like @Klagio wanted some sort of toggle - turning LOCK on turns A on and B off, while LOCK off turns A off and B on. This would be super easy to accomplish with a template switch and a couple lines of automation.

Klagio, 2019-04-01:
> It kinda sounds like @Klagio wanted a master switch - turning LOCK on turns A on and B off, while LOCK off turns A off and B on. This would be super easy to accomplish with a template switch and a couple lines of automation.

I have two (momentary) switch 

switch.open_gate
and
switch.close_gate

wish to have a 
lock.gate

lock would trigger switch.close_gate
unlock would trigger switch.open_gate