esphome/feature-requests #130: [Feature Request] Add Light Level Sensor - LM393 support

davedan, 2019-03-21:
**Describe the problem you have/What new integration you would like**

Request to include Light Level Sensor - LM393 component in the supported devices.

**Please describe your use case for this integration and alternatives you've tried:**

**Additional context**

OttoWinter, 2019-03-22:
The LM393 is just a voltage comparator. When the voltage is above a certain level, it outputs a digital HIGH, else LOW. A gpio binary sensor perfectly represents this.

closed 2019-03-22 by OttoWinter

davedan, 2019-03-23:
The LM393 has a simple photo-resistor light sensor that has both analog and digital outputs. The digital output has a trim potentiometer that can be used to set a trigger light level.

So on the D0 you'll have as you explain HIGH/LOW but in the A0 you receive the amount of light in lumens.

OttoWinter, 2019-03-23:
Yes, and for the light readings (given as an analog voltage) use ADC sensor as seen in TEMT6000 example