esphome/feature-requests #135: Cookbook/Support for Sonoff iFan02

rdbahm, 2019-03-24:
**Describe the problem you have/What new integration you would like**

I use several Sonoff iFan02s in my home automation setup (currently using Tasmota) and would love to get them working with esphome. I'd love to have a cookbook for the iFan02.

**Please describe your use case for this integration and alternatives you've tried:**

These devices run several fans in my house, but I'm hesitant to make many tweaks to them while they're installed, as reflashing them on a ladder via serial if something goes wrong is........ difficult, to say the least!

**Additional context**

There are a few tricky parts about these devices - they have two microcontrollers in them - the second microcontroller some of the actual relay control, so it operates rather differently than many other Sonoff-like devices. In addition, there are some odd quirks to the device - the ESP boots faster than the second microcontroller, so commands sent right after boot (such as setting a light level) won't be obeyed (see [this issue](https://github.com/arendst/Sonoff-Tasmota/issues/3530) on Tasmota for more info).

workingmanrob, 2019-04-03:
Also interested in this cookbook.  ifan02 is my last remaining mqtt device and I would love to retire mqtt.

quazzie, 2019-04-04:
I have not made a cookbook from it but i manager to get the fan working with updates from included remote.
Check https://gist.github.com/quazzie/09ee3ef2c419ecbcf979a7410062481b

quazzie, 2019-04-04:
https://github.com/esphome/esphome-docs/pull/218

workingmanrob, 2019-04-06:
Working for me.  Thanks Quazzie!!  And Otto too of course..

closed 2022-07-01 by nagyrobi