esphome/feature-requests #138: Thermostat / PID

Tuckie, 2019-03-25:
**Describe the problem you have/What new integration you would like**

Requesting a thermostat / PID control that would allow for specifying a sensor, an output pin, and a setpoint. Bonus points for a variety of customization options such as:

* Minimum duty cycle
* Pwm vs binary output
* Auto tuning vs Kp, Ki, Kd
* Open door (lid) detector (See HeaterMeter below)

**Please describe your use case for this integration and alternatives you've tried:**

Building in support for a thermostat or PID controller would allow for local control loop on a device that would not be dependent upon home assistant or the network. This would greatly increase the reliability and safety of the device.

**Additional context**

I have a few different uses for this. First and foremost I would like to use this for controller the blower fan on my charcoal BBQ. I would like to create a much simpler version of the [HeaterMeter](https://github.com/CapnBry/HeaterMeter).

I can also see this being used on a smart switch plugged into a dumb heater fan with a simple thermometer added.

esnil, 2019-06-02:
Love the idea! Especially the part where it becomes independent of connectivity to home assistant. Would allow for a much more robust setup where network and HA-server is not critical for operation.

OttoWinter, 2019-06-02:
FYI, part of this already exist with the bang_bang climate controller. With that you can have a component that accurately keeps some value in a given range. It's not PID but very close.

esnil, 2019-06-03:
> FYI, part of this already exist with the bang_bang climate controller. With that you can have a component that accurately keeps some value in a given range. It's not PID but very close.

Excellent! 
That will do perfectly for my upcoming project! (Still support the idea of a proper PID regulator though) :)

ifujis, 2019-06-07:
That wold be great having a thermostat with a PID controller and PWM output. My use case is a fan coil based AC that has a variable speed blower (PWM controlled) and variable actuator (a valve to control chilled water flow though a radiator). 
It could also have different modes such as:
- Norma cooling - a valve opens proportionally to blower speed
- Drying mode - a valve is fully open while blower speed is low

commander-chris, 2019-06-18:
+1 for a PID controller

blademckain, 2019-07-15:
+1 for a PID controller
the bang_bang controller is very useful but in some cases having a PID function is essential!
For example, I would need a PID function to be associated with a photoresistor to increase / decrease the brightness of some dimmable LEDs.

It should also be fairly simple to integrate already written PID functions. For example
https://playground.arduino.cc/Code/PIDLibrary/

Plx add a generic PID controller!!!

albasee, 2019-08-31:
+2 for a PID controller!

Nicomagno, 2019-10-16:
Yeah, I tried to edit the generated firmware in Arduino to add pid but I don't know enough to get it to work. I would be great to have it in esphome

ifujis, 2019-12-22:
Hi Otto,
Thanks a lot for the pid controller. I've been testing it in my apartment with proportional fan. all good so far. It keeps the temperature more stable now +-0.1C.
Is there a way to attach two outputs to it (e.g. for a water valve too) with a coefficient between the outputs?

pierreh, 2020-03-24:
I was wondering when can we have the PID climate controller available in the official release? I am planning to implement a heater with a PWM controlled fan.

maximoei, 2020-04-29:
Current PID only supports Float outputs, would be great if Binary could also be supported

glmnet, 2020-07-25:
This FR is closed by https://github.com/esphome/esphome/pull/885 if you want something else please open new PR.

closed 2020-07-25 by glmnet