esphome/feature-requests #139: Arduino (Atmega328p) as port expander

glmnet, 2019-03-26:
**Describe the problem you have/What new integration you would like**

Using ESP8266 has not many IO ports, I have some automation done with Arduinos pro mini which I'm upgrading with ESP8266 connectivity, there I am using many IO pins Arduino has available

**Please describe your use case for this integration and alternatives you've tried:**

If you need more IO ports than the available in the esp8266, using a port expander.
I see there are already two port expanders available, I tried an arduino behave as a PCF8574 (actually PCF8575) and got some success, however there where some issues I could not fix.
I did look at the newer port expander, the MCP23017 but not much, as there were other issues/features I'd like to use and this was going to suffer the same problems.

**Additional context**

(When I say Arduino I'm talking about the Atmega328p mostly)
Setting an output in Arduino is different than setting input pull ups, the port expander (PCF8574 at least) seemed to work that way, I don't know too much about electronics but seems like the INPUT_PULLUPs resistors in arduino are 20K and the outputs source up to 40mA current. I really had issues, also with inverted outputs, etc. It ended up working but I had to fix in Arduino setup code which PINs where INPUT_PULLUP and which were OUTPUTs. Also had to add a delay() a the beginning of the setup() in esphome program, as I2C setup was failing.
So I was thinking making a custom port expander using my already built stuff, and had some success too. The good thing is I can setup any pin as I want, INPUT, INPUT_PULLUP and OUTPUT, the ESPHome will setup the pins at the beginning with a special I2C message for every configured port. These are 1 byte send and no byte replied.
Since arduino has 0 to 13 pin digital input, I have 14 available IO ports, and A0 to A7 (except A3, and A4 used for I2C) can also be used as digital intput (A0 to A5) and A0 to A7 obviously as analog inputs. I did implement nothing about analog GPIO yet.
Another Issue I had to solve is Arduino boot time, ESP8266 boots quickly and checks the I2C bus, Arduino is not ready, the check fails and nothing works, so I modified the custom setup of the Arduino port expander to query the I2C for up to 3 seconds until it finds the Arduino it goes on, or otherwise set the component as failed. The ugly thing here besides taking a bit longer to boot is that I2C logs like 10 lines of I2C failure messages.
The reading is done only once per ESPHome loop, I read all the ports in 3 bytes with a I2C query, then keep this value and answer all digital_read queries from components using the cached value. It is reset in the loop event of the port expander component.
Writes are handled with 1 byte message (no ack byte needed), using 3 bits for "command" and 5 bits for port number)
The branch is on my fork (ugly code), I'm opening this feature request for feedback about adding it to the core, it should have a companion arduino code.
I'd like to add the ability to read analog inputs, e.g. analog temperature sensor, Photo resistors, etc, and writing analog outputs (PWM) dim leds, etc. I don't know how easy this can be. 
The arduino code is using less than 10% of the arduino capacity, so other stuff can be added there, like servo library, maybe a IR transmitter that does not get noise from wifi, interrupt counter for an encoder, etc. This would make it way more complicated for me as I don't know much about the esphome-core yet.

redskinhu, 2019-03-26:
Hello

Are you thinkig about like this.?
https://www.letscontrolit.com/wiki/index.php?title=ProMiniExtender

Bye
T

glmnet, 2019-03-26:
@redskinhu yes, actually I got inspired by it.

OttoWinter, 2019-03-27:
No, this goes into the custom component bin - sure it can be shared in cookbook entries and so but I'm absolutely sure some people will want to do more than just port expanding, like you say PWM output and so on. esphome will never be able to support all of that (neither should it be in the core code base). This component would be by design supposed to be customized by the user -> custom component.

closed 2019-03-27 by OttoWinter

glmnet, 2019-03-27:
Thanks for replying.
The problem I have and why I didn't use a custom component is because I did not find the way to register GPIOs with it, I only found documentation on how to create custom sensors, it looks ok for that scenario.

With my code on the Arduino port expander I can do things like this:

```
arduino_port_expander:
  - id: 'ape'
    address: 0x08
  
output:
  - platform: gpio
    id: gpio_relay_1
    pin:      
      arduino_port_expander: ape
      number: 7
      mode: OUTPUT
      inverted: True

light:
  - platform: binary
    id: spot_puerta
    name: "Spot Puerta"
    output: gpio_relay_1

binary_sensor:
  - platform: gpio
    name: "Switch Comedor Puerta 1"
    internal: true
    pin:
      arduino_port_expander: ape
      number: 2 
      mode: INPUT_PULLUP
    on_state:
      then:
        - light.toggle: spot_puerta

```
Is there a way I could do something like this with custom components? Could custom components being added with a base class or something to allow them to have access to this gpio?

e.g.

```
arduino_port_expander:
  - id: 'ape'
    address: 0x08
  
output:
  - platform: gpio
    id: gpio_relay_1
    pin:      
      **custom_component: ape**
      number: 7
      mode: OUTPUT
      inverted: True
```

How otherwise could I achieve something like this?

OttoWinter, 2019-03-27:
> The problem I have and why I didn't use a custom component is because I did not find the way to register GPIOs with it, I only found documentation on how to create custom sensors, it looks ok for that scenario.

See MCP23017 integration as an example. You need to create GPIOPin subclasses.

I think there's a misunderstanding. All esphome components are in a way custom components - they just implement the base components.

> Is there a way I could do something like this with custom components? Could custom components being added with a base class or something to allow them to have access to this gpio?

custom components cannot interact with YAML directly, for that you need to write a python custom component counterpart too.