esphome/feature-requests #140: Cookbook/Support for Sonoff SC

hemantkamalakar, 2019-03-26:
**Describe the problem you have/What new integration you would like**
I managed to migrate my Sonoff basic devices from Tasmota to esphome. But facing issues with Sonoff sc. Initially I flashed sonoff sc using OTA. But can't figure out how to define the sensors available in Sonoff SC in esphome yaml file. I'd really love to have a cookbook for Sonoff SC.

**Please describe your use case for this integration and alternatives you've tried:**
I have tried following sensor definition in my esphome yaml file for DHT11 sensor.
`sensor:
  - platform: dht
    model: DHT11
    pin: 10
    temperature:
      name: "Sonoff Temperature"
    humidity:
      name: "Sonoff Humidity"
    update_interval: 10s`

This is what i found in the logs.

`14:01:26][W][sensor.dht:140]: Requesting data from DHT failed!
[14:01:26][W][sensor.dht:076]: Invalid readings! Please check your wiring (pull-up resistor, pin number).` 

**Additional context**
Sonoff SC is a sensor station that packs a DHT11 temperature and humidity sensor, a GM55 LDR, a microphone with an amplifier circuit and a Sharp GP2Y1010AU0F dust sensor. Here is the schematic of sonoff SC. [https://www.itead.cc/wiki/images/f/f7/Sonoff-SC-Schematic.pdf](url).

glmnet, 2019-03-27:
Why pin 10?

nickrout, 2019-03-27:
The DHT 11 is shown as connectiong to pin 10 on an MCU, but if you look carefully it is a ATMega328P. There two MCU's in this device, the ESP8266 and the ATMega328P. The sensors are connected to the latter. It is therefore a bit different to just flashing to the esp8266. There is a good analysis here http://tinkerman.cat/sonoff-sc-with-mqtt-and-domoticz-support/

Bigmanekb, 2019-05-08:
So, i see that SC is not fully supported yet ? It's sad. I migrated to ESPHome from Tasmota only because i thought that SC is fully supported.

MimbaMonkeyHouse, 2019-08-12:
Would love to see this happening too! I just ain't got the skills...

bastshoes, 2019-09-06:
Sonoff SC have two boards inside. ESP and Atmel. ESP responsible only for outside communication. All sensors connected to Atmel board. Two boards communicating via serial. If you can implement communication protocol inside ESP you'll able to publish data anywhere.