esphome/feature-requests #142: Cookbook for Sonoff Dual (*not R2*) with serial interface

daenny, 2019-03-28:
**Describe the problem you have/What new integration you would like**

Properly integrate Sonoff Dual version 1. Sonoff Dual v1 uses a serial interface to control the relays. 

I have created a workaround which works for my case, since I need interlocking for setting the fan speed. But I think there should be a better way. 

**Please describe your use case for this integration and alternatives you've tried:**
```yaml
substitutions:
  devicename: ventilation
  upper_devicename: Ventilation
  ip_end: "220"

esphome:
  name: ${devicename}_switch
  platform: ESP8266
  board: esp01_1m
  includes:
    - sonoff_dual_out.h

wifi:
  ssid: 'xxx'
  password: 'xxx'
  manual_ip:
    static_ip: "192.168.2.${ip_end}"
    gateway: 192.168.2.1
    subnet: 255.255.255.0

uart:
  tx_pin: GPIO1
  rx_pin: GPIO3
  baud_rate: 19200

# Enable logging
logger:
  baud_rate: 0

# Enable Home Assistant API
api:
  password: 'xxx'

ota:
  password: 'xxx'

switch:
- platform: custom
  lambda: |-
    auto sonoff_out1 = new SonoffOut1();
    App.register_component(sonoff_out1);
    return {sonoff_out1};
  switches:
    - name: "${upper_devicename} Mode Medium Internal"
      id: relay1
      internal: True
- platform: custom
  lambda: |-
    auto sonoff_out2 = new SonoffOut2();
    App.register_component(sonoff_out2);
    return {sonoff_out2};
  switches:
    - name: "${upper_devicename} Mode High Internal"
      id: relay2
      internal: True
- platform: template
  name: "${upper_devicename} Mode Medium"
  lambda: |-
      if (id(relay1).state) {
        return true;
      } else {
        return false;
      }
  turn_on_action:
    - switch.turn_off: relay2
    - switch.turn_on: relay1
  turn_off_action:
    - switch.turn_off: relay1
- platform: template
  name: "${upper_devicename} Mode High"
  lambda: |-
      if (id(relay2).state) {
        return true;
      } else {
        return false;
      }
  turn_on_action:
    - switch.turn_off: relay1
    - switch.turn_on: relay2
  turn_off_action:
      - switch.turn_off: relay2
- platform: restart
  name: "${upper_devicename} Switch Restart"

status_led:
  pin:
    number: GPIO13
    inverted: yes

# Example configuration entry
web_server:
  port: 80
```

```c++
#include "esphome.h"
using namespace esphome;

class SonoffOut1 : public Component, public switch_::Switch {
 public:
  void setup() override {
    // This will be called by App.setup()
  }
  
  void write_state(bool state) override {
    Serial.write(0xA0);
    Serial.write(0x04);
    if (state)
      Serial.write(0x05);
    else
      Serial.write(0x00);
    Serial.write(0xA1);
    Serial.write('\n');
    Serial.flush();
    publish_state(state);
  }
};

class SonoffOut2 : public Component, public switch_::Switch {
 public:
  void setup() override {
    // This will be called by App.setup()
  }

  void write_state(bool state) override {
    Serial.write(0xA0);
    Serial.write(0x04);
    if (state)
      Serial.write(0x06);
    else
      Serial.write(0x00);
    Serial.write(0xA1);
    Serial.write('\n');
    Serial.flush();
    
    publish_state(state);

  }
};
```

**Additional context**
This is probably similar for the Sonoff 4CH.

MarekLe, 2019-05-20:
@daenny could you chelp me to change your code? I wonna to create code for sonoff dual (Touch). Your code works but it has some kind of 'interlocking'. When I turn on relay1 - relay2 is turned off automatically. How to change this to get working it independently? I wont to control relay1 and relay 2 separatly.

Sorry for bad english....

OttoWinter, 2019-05-25:
Yes there is a better way, using the uart switches you can send the signals without leaving YAML.

filmgarage, 2019-08-27:
What would the better way look like? I am trying to put a 4 relay switch together as well. It needs three relays to support interlocking and communicates thorough serial as like the Sonoff... Standard control for interlocking is not suitable for these kinds of relays right?

OttoWinter, 2019-08-27:
@daenny Closing as now `uart.write` action exists (which makes this fully YAML-able)

@filmgarage I don't see how your question is related to the feature request. But I also don't really understand the question fully.

closed 2019-08-27 by OttoWinter