esphome/feature-requests #144: RDM6300: add on_tag automation

balloob, 2019-03-29:
**Describe the problem you have/What new integration you would like**

I would like to configure an `on_tag` automation for the RDM6300 component. A similar feature is available for the very similar pn532 component.

One thing that might impact this feature is that the RDM6300 triggers multiple times in case the card is held for a longer period of time in front of the antenna. I need to be _very_ quick and still get at least 2 triggers.

If such a feature would be added, it would need a cooldown to test for a card not being on the scanner for X seconds before allowing to trigger again. 

**Please describe your use case for this integration and alternatives you've tried:**

My goal is to forward all scanned tags to Home Assistant as events when #89 is implemented.

My ultimate goal is to integrate https://github.com/maddox/magic-cards/ more tightly into Home Assistant using ESPHome.

**Additional context**

❤️

OttoWinter, 2019-04-25:
Added in dev

closed 2019-04-25 by OttoWinter

SubOrbit, 2019-05-21:
Docs for the on_tag action should say:
`  on_tag:
    then:
      - mqtt.publish:
          topic: rdm6300/tag
          payload: !lambda 'return uint32_to_string(x);'`

Current `payload: !lambda 'return x;'`` results in compile error...