esphome/feature-requests #145: Cookbook for MPDMv7.5 with DAC (MPC4726)

daenny, 2019-03-29:
**Describe the problem you have/What new integration you would like**

I created a MPC4726 DAC output component tailored to the MPDMv7.5. I already had done something similar for Tasmota (https://github.com/arendst/Sonoff-Tasmota/pull/3015).
Maybe that is useful for some other people as well.

```cpp
#include "esphome.h"
using namespace esphome;

class DacOut : public Component, public output::FloatOutput {
 public:
  void setup() override {
    // This will be called by App.setup()
    Wire.begin();
    pinMode(16, OUTPUT);

  }

  void write_state(float state) override {
    //first check if the general output should be enables
    if (state == 0.0) {
        digitalWrite(16, 0);
    }
    else {
        digitalWrite(16, 1);
    }
    // state is the amount this output should be on, from 0.0 to 1.0
    // we need to convert it to an integer first
    int value = state * 1024;
    //Map to 2100-3350, values by trial and error, may depend on used light. I am using LEDs. 
    uint16_t output = (uint16_t) map((1024-value), 0, 1024, 2100, 3350);
  
    Wire.beginTransmission(0x60); //address of DAC
    Wire.write(0x40); //write data to DAC
    Wire.write(output >> 4);                   // Upper data bits          (D11.D10.D9.D8.D7.D6.D5.D4)
    Wire.write((output & 15) << 4);            // Lower data bits          (D3.D2.D1.D0.x.x.x.x)
    Wire.endTransmission();

  }
};
```

yaml config:
```yaml
substitutions:
  devicename: ikea_stand
  upper_devicename: Ikea Stand
  ip_end: "207"

esphome:
  name: ${devicename}_light
  platform: ESP8266
  board: esp01_1m
  includes:
    - dac_out.h

i2c:
  sda: GPIO4
  scl: GPIO5
  scan: True
  frequency: 400kHz

wifi:
  ssid: 'xxx'
  password: 'xxx'
  manual_ip:
    static_ip: "192.168.2.${ip_end}"
    gateway: 192.168.2.1
    subnet: 255.255.255.0

# Enable logging
logger:

# Enable Home Assistant API
api:
  password: 'xxx'

ota:
  password: 'xxx'

output:
- platform: custom
  type: float
  lambda: |-
    auto my_custom_float_output = new DacOut();
    App.register_component(my_custom_float_output);
    return {my_custom_float_output};
  outputs:
    id: dac_out

# Example configuration entry
light:
  - platform: monochromatic
    name: "${upper_devicename} Light"
    output: dac_out
    default_transition_length: 0.1s

switch:
  - platform: restart
    name: "${upper_devicename} Light Restart"

# Example configuration entry
web_server:
  port: 80
```

**Please describe your use case for this integration and alternatives you've tried:**

**Additional context**

OttoWinter, 2019-03-30:
Sure, that looks like a simple integration. However, for it to get into the esphome codebase it will need to use the esphome abstractions (GPIOPin, I2CDevice

bpmittal, 2019-04-21:
is this the ac dimmer working with esphome ? can i use it to control my ac fans ?
does esphome supports directly ?

LeandroSPereira, 2019-11-30:
Hello Daniel, how are you?

I am researching a way to use my MPDMv4.1.3 module to dimmer some of my home environments, and during my research I found your github with some great information, but my lack of IOT knowledge is making things difficult, I have an ESP8266 Lolin V3, and would like to use with ESPHOME to control this module, if you have some free time at some point, could you help me in this endeavor? Thanks in advance for the material available, I thought I would never be able to use these modules.

Thank you very much

daenny, 2019-11-30:
Are you using the DAC? Otherwise it's "just" a PWM light in Pin 5 if I read it correctly here
http://esp8266-projects.org/2016/12/espeasy-p2-domoticz-rpi-mpdmv4-ac-dimmer-setup-example/

LeandroSPereira, 2019-11-30:
I don't understand this DAC issue very much, I would like to control the board with a Nodemcu (Lolin v3) based on ESPHOME for use in Home Assistant.

closed 2021-11-10 by daenny