esphome/feature-requests #146: Implement BLE subscribe to Notify

marcellus00, 2019-03-31:
**Describe the problem you have/What new integration you would like**
Current BLE implementation is limited to Tracker/Beacon/Xiaomi sensors devices and scenarios.
It's impossible to setup a sensor, that will react to other BLE server's Notify event. I think such feature will make the BLE component much more flexible.

**Please describe your use case for this integration and alternatives you've tried:**
I'm trying to make a Bluetooth LE button based on ESP32. It should be battery powered so wifi is out of the picture. 
What I've done: a button that will emit a BLE notify event 
What I'm looking for: a solution that will listen for a button and automatically integrate it into Home Assistant. 
What I've tried: 
1) Node-red with ble node. Never made it to recognize my bt adapter. In docker it required compiling alpine packages, but dependencies couldn't be resolved. I've tried to use a plain node-red install on ubuntu, but that time ble node itself refused to work.
2) Standalone python program. It works, but it lacks proper home assistant integration.

**Additional context**
There's a registerForNotify() usage example at https://github.com/nkolban/esp32-snippets/blob/master/Documentation/BLE%20C%2B%2B%20Guide.pdf
I'm currently studying esphome sources and would like to contribute, but C++ is not my proficiency and I'm not sure that I can pull this off, at least the right way.
My "button" part: https://github.com/marcellus00/ESP32_BLE_Button

marcellus00, 2019-03-31:
I'm really sorry, but Custom Component is indeed all I need to implement such functionality. I've seen it before but somehow it slipped my mind. Thanks, rubber duck!

marcellus00, 2019-03-31:
I'm really sorry for this frantic issue handling, but after the night of comprehending the sources I came to the conclusion that issue is indeed present:
1) esp32_ble_tracker component implements an exquisite request handling system using semaphores, so every ble device requests won't interfere with each other; therefore all new ble scenarios should be implemented using this system. (On the side note, this appears to me as a kinda tight coupling and may be a bottleneck in the future development.)
2) Implementing ble interaction in custom sensor for more than one ble server appears to be challenging and for sure will interfere with esp32_ble_tracker. (Please, can someone explain me how to declare new config variables for custom components?)

OttoWinter, 2019-04-01:
Yeah ble tracker needs to be untangled into a more modular approach at some point. Probably just adding a callbackmanager that passes out the `ESPBTDevice` instances should be fine.

However, since I don't see too many other BLE adv devices being added in the near future this is low priority.

OttoWinter, 2019-04-25:
The API for BLE tracker has been rewritten a bit to allow this in dev - see `ESPBTDeviceListener`.

closed 2019-04-25 by OttoWinter