esphome/feature-requests #147: AirThings BLE Radon Sensor

brandond, 2019-04-02:
https://airthings.com/raspberry-pi/

**Please describe your use case for this integration and alternatives you've tried:**
Support for reading radon levels and other sensors from device via BLE.

**Additional context**

https://community.home-assistant.io/t/air-quality-monitor-radon-meter-airthings-wave-plus/102836

hteggi, 2019-12-02:
I would also like to see this in ESPhome. I picked up a Wave Plus a few days ago and used this:  https://github.com/gkreitz/homeassistant-airthings custom component that works well for some time. 

But as with my Xiaomi flora and bluetooth temperature sensor it stops working after some hours on the Raspberry Pi 3 running hassos.

After I changed to ESPhome for the Xiaomi bluetooth devices no problem so thanks for this great addon.

areeeee, 2020-01-22:
I have a Airthings Wave Plus and would like to help if someone has the skills to make this.

debsahu, 2020-02-18:
ESP32 based readings from wave is possible https://github.com/sabeechen/AirthingsMQTT Unrelated: I combined this with Xiomi Flora code to get it going for me, and it is not esphome based.

kressir, 2020-08-17:
Yes - please add.  Please also include the illumination and acceleration as Marty has done here: https://github.com/custom-components/sensor.airthings_wave

I have an Airthings Wave (non Plus) that I can help test with.

ejdan, 2020-09-26:
I have a 2nd gen Wave so I am very interested in this. I managed to modify and run this https://github.com/sabeechen/AirthingsMQTT on my ESP32 and it seems to work fine. Unfortunately it is written using a different Bluetooth library (Arduino BLE vs Espressif Bluedroid).

I ran in to problems when trying to make it into an ESPHome component. The problem is that all the BLE sensor components in ESPHome seems to be passive. From what I can tell you need to actively request data from the Wave.

Is there an example of an ESPHome BLE component which is actively requesting data at a fixed interval? I found the BLE Beacon component but is it possible to run both Tracker and Beacon at the same time?

ejdan, 2020-11-12:
Looks like there is some progress here:
https://github.com/esphome/esphome/pull/1177

When this has been merge I think it will be much easier to implement Airthings.

jeromelaban, 2021-04-03:
I've taken a stab at implementing specific support for this device in https://github.com/esphome/esphome/pull/1656. It adds a device scanner and sensors.

sylvinus, 2021-07-06:
Hey @jeromelaban, funny seeing you here :)

I just tested this successfully with a 1st gen Wave Plus (Model 2930xxxx).

Only issue I had was https://github.com/esphome/esphome-docs/pull/1081/files#r664574473

Great work!

closed 2021-08-31 by jesserockz