esphome/feature-requests #148: Support to messure more than 54612 Lux with the BH1750

rradar, 2019-04-02:
**Describe the problem you have/What new integration you would like**

The esphome integration of the BH1750 Lux Sensor only allows to measure a maximum of 54612 Lux (in my tests). There is a possible to (software wise) measure more than 100000 Lux with this sensor at the costs of Measurement Time (MTreg) which can go up to a maximum of around 500 ms.

**Please describe your use case for this integration and alternatives you've tried:**

Like to measure outside with sun which easily exceeds 54612 Lux

**Additional context**

There is a library and a little bit of context on this (german only) website:

> Virtueller Modus:
> RESOLUTION_AUTO_HIGH: Die Werte im MTreg ('Measurement Time'-Register) werden je nach Helligkeit automatisch so angepasst, dass eine maximalmögliche Auflösung und Messbereich erziehlt werden. Die messbaren Werte fangen von 0,11 lx und gehen bis über 100000 lx. (ich weis nicht, wie genau die Werte in Grenzbereichen sind, besonders bei hohen Werte habe ich da meine Zweifel. Die Werte scheinen jedoch weitgehend linear mit der steigenden Helligkeit zu wachsen.) Auflösung im Unteren Bereich ca. 0,13 lx, im mittleren 0,5 lx, im oberen etwa 1-2 lx. Die Messzeiten verlängern sich durch mehrfache Messungen und die Änderungen von Measurement Time (MTreg) bis max. ca. 500 ms.

http://s6z.de/cms/index.php/homeautomation-homecontrol/infrastruktur/aktorensensoren/offene-standard-systeme/i-c/15-umgebungslichtsensor-bh1750

rradar, 2019-05-06:
Actually the source code already exists and is licensed under LGPL

Code can be found here: https://github.com/hexenmeister/AS_BH1750

It does also include a 'auto power down' beside supporting the measurement range from 0,11 to 100.000 lux.

rradar, 2019-10-23:
Hey @OttoWinter sorry to interrupt but as I'm seeing you getting ready for the 1.14 release I wonder how hard this one here is to implement.

I'm driving my whole water and water-heating (solar) system with esphome and I'm just hardly lacking the feature to measure the radiated lux on the solar panel. The sensor is already in place but maxes out on 54612 lux.

OttoWinter, 2019-10-24:
> which can go up to a maximum of around 500 ms in this case (which is not bad at all!).

It would actually be the opposite - shorter measurement times allow for a higher range.

rradar, 2019-11-04:
Could their be something tested with the new code already @OttoWinter ? I have a couple of esp's with bh1750 including one towards the sky and would be happy to contribute with testing.

esphome-dev is setup and ready but I don't know if I need to add any setting to the bh1750 component in the yaml to get shorter measurement times for higher range or if it is done automatically?

OttoWinter, 2019-11-04:
You can test it by downloading [this folder](https://github.com/esphome/esphome/tree/03cca03e204638e05d5d7ad3446b7a957ea355a6/esphome/components/bh1750) and placing the files under `custom_components/bh1750` in your ESPHome config dir.

rradar, 2019-11-05:
I did compile esphome 1.14.2 with the [bh1750 custom component](https://github.com/esphome/esphome/tree/03cca03e204638e05d5d7ad3446b7a957ea355a6/esphome/components/bh1750) included and flashed it via ota (patching the running 1.13.6)

Still it maxes out at 54612.5.

[image]

I didn't see in the compiling logs that the custom component was used but as a folder `__pycache__` was created inside the bh1750 folder while compiling it looks like it was accepted.

OttoWinter, 2019-11-05:
Ah sorry, forgot to mention: you need to supply the new measurement_time via config - the default is still the old one.

```yaml
sensor:
  - platform: bh1750
    name: "${upper_devicename} Panel Brightness"
    address: 0x23
    # min: 31, default: 69, max: 254
	measurement_time: 31
```

rradar, 2019-11-06:
Confirmed working @OttoWinter !

[image]

Is their any fixed upper/lower limit when having the `measurement_time` set to `31`/`254`? I will try to update the docs correspondingly!

Thank's so much! This is a great improvement for this component! :tada:

OttoWinter, 2019-11-07:
The max value for `31` would be 121556 lx and for `254` would be 14835 lx - as with any exposure time, longer exposure is good for low-light, short exposure is good for sunny scenes.

rradar, 2019-12-15:
Updated docs: https://github.com/esphome/esphome-docs/pull/440

martin3000, 2020-03-25:
will this go into the normal release?

rradar, 2020-03-25:
It need's to @martin3000! But first the [pull request #997](https://github.com/esphome/esphome/pull/997) needs to be fixed to pass the tests.

martin3000, 2020-03-25:
@rradar how do these automated tests work?

rradar, 2020-03-30:
@martin3000 travis does all the work:

[image]

He also says what to change to pass the test. In this case it's easy and some spaces in the right place should make the tests pass in the next run:

[image]

closed 2020-05-29 by glmnet