esphome/feature-requests #149: Pulse Width Sensor

timpur, 2019-04-02:
**Describe the problem you have/What new integration you would like**

To measure the pulse width not the pulse count. ie the length of time each pulse is.

**Please describe your use case for this integration and alternatives you've tried:**

I'm creating a fully custom wind sensor, that uses a hall sensor and a magnet to measure the speed of rotation. This will hopefully tell me the speed of the wind (once calibrated).

**Additional context**

The reason i want to use pulse width and not pulse count, is i want to be able to calculate the instantaneous velocity, not map rpm to speed. I believe this will allow me to know the min max and average speed for any given period (ie 60s). Which is handy for wind speed, as rpm only tells me that average speed for a given period.

I also think this might be useful to interpret signals like PWM, where the pulse width is important. So maybe we can come up with a interface that is most useful for everyone.

I am also willing to help out with the code and can lead the charge, though i need to spend more time yet understanding esphome code ... still new :P

Keen for your thoughts. Thanks

timpur, 2019-04-05:
I did not see this https://esphome.io/components/sensor/duty_cycle.html, though it doesnt do what i need .. but might be able to expand on it or something.

OttoWinter, 2019-04-25:
Has been added in dev - docs are not written yet though.

closed 2019-04-25 by OttoWinter

timpur, 2019-04-25:
Carnt seem to find it in code ?

OttoWinter, 2019-04-26:
See https://github.com/esphome/esphome/tree/dev/esphome/components/pulse_width

timpur, 2019-04-26:
Thanks, I build the component but it also had min max and pwm %. Do you think any of that is worth adding to this component ?

OttoWinter, 2019-05-01:
> Thanks, I build the component but it also had min max and pwm %.

min/max are template sensors IMO.
PMW % is duty_cycle.

ESPHome is intentionally very modular. Components do one thing and one thing only. The user can pick and combine them afterwards.

timpur, 2019-05-01:
Normally i'd agree, but this approach wont meet my requirements. I need to get the single max and min pulse time for a given period, say my period is 1 sec, there could be a pulse that is 10ms long and one 30ms long but the rest are 20ms, so i need to know the outliers which this component wont let me do even with templating....

I could reduce the period of the component and then add filters, but this would still only be a hack (close approx.)... while i need the outliers not the outliers of an average result for a period. 

if that made any sense.

Have a look at my implementation:
https://github.com/timpur/esphome-core/blob/e399adfc3e10d08c23d3549f4fa18ef237bd1216/src/esphome/sensor/pulse_width.cpp#L46
https://github.com/timpur/esphome-core/blob/dev/src/esphome/sensor/pulse_width.h

It records:
* PWM
* Frequency
* For both High and Low side ->avg, min, max

Thus you can see most details about the PWM wave from.