esphome/feature-requests #151: Replace Total Daily Energy Sensor with general component for integration of signals over time

mkuoppa, 2019-04-03:
**Describe the problem you have/What new integration you would like**

Total Daily Energy Sensor can already today be used for general integration of signals over time. Why not make a general component instead. 

**Please describe your use case for this integration and alternatives you've tried:**

My use case is a rain gauge sensor (pulse counter) that reports mm/minute and it seems to work to use the Total Daily Energy Sensor to aggregate the amount of rain during 24h.

**Additional context**

nickrout, 2019-04-03:
From my personal perspective (non developer, merely observer) home assistant can do this, and we shouldn't burden our little esp's with holding long term data. https://www.home-assistant.io/components/sensor.integration/

mkuoppa, 2019-04-03:
I agree, home assistant have a wide range of components that can do this and more. But I think there are use cases were it would be more or less necessary to do it in the esp. Sending a lot of data often to HA is not always the best alternative. And since there already is a component basically doing this...

nickrout, 2019-04-03:
True about the data. But does the total get saved in an ESP if it is unplugged suddenly?

OttoWinter, 2019-04-09:
>  But does the total get saved in an ESP if it is unplugged suddenly?

Yes it does.

> Sending a lot of data often to HA is not always the best alternative.

Well the overhead of doing so is tiny - especially since the ESP and HA ping each other anyway every 30s to keep the connection alive.

zewelor, 2019-04-13:
Whats the interval, for writing that state into flash ? Won't it wear flash memory, when total_daily_energy is used ?

OttoWinter, 2019-04-15:
> Whats the interval, for writing that state into flash ? Won't it wear flash memory, when total_daily_energy is used ?

Interval is update rate of the sensor (each time it publishes a new value, state is rewritten). I guess the docs could receive a note about that. However, I've been using this on an ESP32 with nvs for more than 2 years now with an update rate of a minute and have not seen the flash being worn out yet.

kluszczyn, 2019-04-16:
I second this future request. Also it would be beneficial to add option to set/reset value of sensor (counter) through yaml/lambda.

Good to know that counter is saving it's state to flash every update. In my setup I use this sensor with anemometer, with update rate 1s. I think I need to switch to global variable in this case.

OttoWinter, 2019-04-25:
integration sensor has been added (docs are not written yet though).

closed 2019-04-25 by OttoWinter