esphome/feature-requests #152: Simple UART FloatOutput (With proof-of-concept!)

nekromant, 2019-04-04:
It would be a good idea to provide a simple UART-based output::FloatOutput that can be used directly from config. This would cover some cases when we have a dumb uC that handles PWM output.  

I have created  a simple proof-of-concept how it may look like using Black Board T5 ESP-13 board. 

The protocol in this concept is as simple as: 
- One characted prefix 
- Three bytes of textual data representing 0-255 pwm value. 

What's not covered: 
- More than 8 bits of pwm
- Float pwm output
- The protocol is just a quick hack, something universal like prefix, suffix, data format/endianness might be required. 

P.S. Can help out with C++ part. Unfortunately not familiar with the python part that handles all configuration parsing. 

P.P.S. Sources:
UART FloatOutput component: https://github.com/nekromant/blackboard-STC15/blob/master/ESPHome/stc15pwm.h
Complete yaml file: https://github.com/nekromant/blackboard-STC15/blob/master/ESPHome/bb.yaml
STC15 firmware handling this stuff: https://github.com/nekromant/blackboard-STC15/blob/master/esphome.c

OttoWinter, 2019-04-04:
Nah, this is a custom component and should be distributed as such.

If we make this an official integration, the user is tied to this protocol and can't change it easily (which *will* happen). So by making it a custom component users can very easily go into the source file and change the protocol to suit their needs.

I'm not saying we shouldn't put it in the docs, we should (can inspire other similar projects). But making it an official component would be too restrictive on the protocol. Other option would be an uber-generic protocol, but that's even worse because such a protocol is impossible.

This fits well in the cookbook section.

nekromant, 2019-04-04:
Perhaps you are right. It's difficult to make something uber-generic, so I decided to ask first beforehand.

closed 2019-04-04 by nekromant