esphome/feature-requests #155: SenseAir S8 CO2 sensor

CiTroNaK, 2019-04-05:
**Describe the problem you have/What new integration you would like**
I would like to use SenseAir CO2 sensor with ESPHome. See https://senseair.com/products/size-counts/senseair-s8-residential/

**Please describe your use case for this integration and alternatives you've tried:**
Same as for other CO2 sensors.

**Additional context**
https://www.letscontrolit.com/wiki/index.php/S8

Thank you very much.

StanislavBolshakov, 2019-05-01:
SenseAir support would be a welcome and highly appreciated addition.

Spirituss, 2019-07-03:
I also look forward its support.

acidquest, 2019-07-17:
i wait support for this bestest from cheap co2 sensor

Spirituss, 2019-07-17:
The S8 device has detailed specifications, so it should not be very difficult to add the support for it. At the moment, the inability to use ESPHome for S8 is the main reason to use Tasmota firmware.

to4ko, 2019-07-28:
looking forward for S8 support in ESPhome

G1K, 2019-07-28:
@OttoWinter , 
This is a very good and frequently used sensor, what is the probability of implementing it in ESPHome ?

datasheet:
http://www.co2meters.com/Documentation/Datasheets/DS-S8-3.2.pdf
http://senseair.senseair.com/products/oem-modules/senseair-s8/

ESPEasy - 2016 [plugin](https://github.com/letscontrolit/ESPEasy/blob/2d6988f78616ea87285129aefca2386b2af07988/src/_P052_SenseAir.ino)
Senseair S8 and K30, K70 
https://github.com/letscontrolit/ESPEasyPluginPlayground/commit/f5102918bcfd4385d2584789286e7da8a1a61ad7

Tasmota - 2017
https://github.com/arendst/Sonoff-Tasmota/commit/7723d52453ebcbbb0ea6c0826bea080f83343214

ESPurna - 2018 
https://github.com/xoseperez/espurna/pull/812/commits/ce10f72a6c4cc66a186d83db987d2e530a1b20fb

KarpushinVA, 2019-07-29:
Please, add support for S8 sensor.

freenetwork, 2019-07-29:
Please, add support for SenseAir.

Bird55, 2019-07-29:
I plan to use this sensor, turn it on in the EspHome

msv5868, 2019-07-29:
Добавьте пожалуйста датчика S8 в EspHome.

Anonym-tsk, 2019-08-27:
Кто SenseAir S8 в ESPHome заказывал?
Проверяйте (ветка senseair): https://github.com/Anonym-tsk/esphome/tree/senseair
Датчика у меня нет, но компилируется )

G1K, 2019-08-27:
> Who SenseAir S8 in ESPHome ordered?
> Check (line senseair): https://github.com/Anonym-tsk/esphome/tree/senseair
> I don't have a sensor, but it compiles )

Tests the firmware successfully, the migration from espeasy one sensor is produced.  Approximately with such config works
[image]

```
uart:
  rx_pin: GPIO13
  tx_pin: GPIO15
  baud_rate: 9600
  
sensor:
  - platform: senseair
    co2:
      name: "Senseair S8 co2"
    update_interval: 60s
```

Thanks @Anonym-tsk

Anonym-tsk, 2019-08-29:
I bought a sensor and tested the code. Everything works perfectly.

SCWhite, 2019-09-03:
I have some question about SenseAir S8 CO2 sensor
it seems that the Automatic Baseline Correction (ABC) function can't be changed

Lefey, 2019-09-03:
ABC can be disabled, as well as calibrate at 400 ppm or 0 ppm. At this moment, you can disable calibration only with UIP manufacturer software, you can found it here https://senseair.com/support/download/software/

SCWhite, 2019-09-09:
@Lefey  Thank you for providing this solution, the software works and shows the CO2 reading.(haven't try to adjust the ABC function yet) 
more specifically saying the ABC function can't change by modbus command. 
(or only could be change when "right after a ABC is done" )
see this post: [](https://github.com/letscontrolit/ESPEasy/issues/759)

This could be a problem in the future.

closed 2019-10-17 by OttoWinter