esphome/feature-requests #158: ESP-DASH (advanced web ui) Integration

rradar, 2019-04-06:
**Describe the problem you have/What new integration you would like**
ESP-DASH, awesome Dashboards (web ui) for our esp which not rely on any internet connection

**Please describe your use case for this integration and alternatives you've tried:**

The web server component is very simple and it would be a dream to get ESP-DASH into the show! :star: :+1: :1st_place_medal: :grin: 

**Additional context**

Demo: https://ayushsharma82.github.io/ESP-DASH

Github: https://github.com/ayushsharma82/ESP-DASH/issues/19

OttoWinter, 2019-04-07:
The web server is purposefully that simple, esphome does not try to replace your home automation hub.

That being said, I'd be open to make the web server UI a bit nicer, though I'm pretty sure esp-dash won't be the solution because it uses a very different architecture from what I see.

I will close this feature request as it's probably not possible to implement it into esphome directly (without making architectural changes). That's not to say that efforts to make the web server interface nicer will be stopped, just that this exact way will not work.

closed 2019-04-07 by OttoWinter

rradar, 2019-04-07:
Okay dokay! Thank's for the information!