esphome/feature-requests #159: MQ-7 Carbon Monoxide Sensor

kgalilio, 2019-04-06:
**Describe the problem you have/What new integration you would like**

I would like to have MQ-7 Sensor added to the ESPHome list of supported sensors. 
**Please describe your use case for this integration and alternatives you've tried:**

The MQ-7 sensor is not supported by ESPHome. It has an arduino library for it, but not available in the supported sensors list in ESPHome.
[image]

[image]

[image]

**Additional context**

rradar, 2019-04-06:
**You can use it already** just use the adc component (give a look at the temt600 cookbook entry for inspiration) for getting the analog values. Also digital out you can use - this is the one which is triggered if a certain threshold is exceeded - you can set the level with the poti - also the red D0 Led will light up.

With the MQ-7 Sensor the hardest part is the proper voltage feeding. It want's a special mix between 1.5v and 5v in a certain interval to give proper results. You can see this in your data sheet you posted. Also the sensor needs to be calibrated - which can be quite a hard task! Good luck!

OttoWinter, 2019-04-07:
Yes, as @rradar said this is already supported as it's an analog sensor. With these integrations that can be implemented with a very short lambda esphome's policy is not to include them as dedicated components.

See TEMT6000 as an example. However, I'm open to adding it as a cookbook example, somebody just needs to do it.

I will close this as it's already possible.

closed 2019-04-07 by OttoWinter

rdehuyss, 2019-04-13:
@OttoWinter how would you then do the power cycling? 

Because for the MQ-7 you need to:
- Apply 5V for 60 seconds, don't use these readings for CO measurement.
- Apply 1.4V for 90 seconds, use these readings for CO measurement.

The TEMT600 is easy as it always requires a constant vcc, this is not the case for the MQ-7

@rradar do you want to team up on this?

kgalilio, 2019-04-13:
> 
> 
> @OttoWinter how would you then do the power cycling?
> 
> Because for the MQ-7 you need to:
> 
>     * Apply 5V for 60 seconds, don't use these readings for CO measurement.
> 
>     * Apply 1.4V for 90 seconds, use these readings for CO measurement.
> 
> 
> The TEMT600 is easy as it always requires a constant vcc, this is not the case for the MQ-7
> 
> @rradar: how did you do the power cycling?
> @kgalilio: do you want to team up on this?
> 
> Basically, I would update the cookbook if I would know how to do it :-)

I certainly want to, but I am a total noob!

I got this code from where I got my sensor from, I hope it helps:

```
Electrical properties:
Input voltage: DC5V power (current): 150mA
DO output: TTL digital 0 and 1 (0.1 and 5V)
AO output :0.1-0 .3 V (relatively clean), the highest concentration of voltage around 4V
Special note: The sensor is powered, needs to warm up around 20S, measured data becomes stable, the sensor fever is a normal phenomenon, because the internal heating wire, if hot is not normal.
 
Test Procedure:
 
Function: This version of the matching test program
Using chips: AT89S52
Crystal: 11.0592MHZ
Baud Rate: 9600
Build Environment: Keil
[Statement] This procedure is used only for study and reference, please specify the copyright and author information!
************************************************** ******************* /
/ ************************************************* *******************
Note: 1, when measured concentrations greater than the set concentration, single-chip IO output low
************************************************** ******************* /
# Include / / library files
# Define uchar unsigned char / / unsigned char macro definition
# Define uint unsigned int / / macro definition unsigned integer
 
/ ************************************************* *******************
I / O definition
************************************************** ******************* /
sbit LED = P1 ^ 0 ;/ / define MCU port P1 first one (ie, P1.0) to indicate the end of the
sbit DOUT = P2 ^ 0 ;/ / define MCU port P2 of the first one (ie, P2.0) for the sensor input
/ ************************************************* *******************
Delay function
************************************************** ******************* /
void delay () / / delay procedure
{
uchar m, n, s;
for (m = 20; m> 0; m -)
for (n = 20; n> 0; n -)
for (s = 248; s> 0; s -);
}
/ ************************************************* *******************
The main function
************************************************** ******************* /
void main ()
{
while (1) / / infinite loop
{
LED = 1 ;/ / off P1.0 port lights
if (DOUT == 0) / / When the concentration is higher than the set value, the function execution condition
{
delay () ;/ / delay interference
if (DOUT == 0) / / determine the concentration is higher than the set value, the function execution condition
{
LED = 0 ;/ / P1.0 port lights lit
}
}
}
}
/ ************************************************* *******************
End
************************************************** ***************** 
```

OttoWinter, 2019-04-15:
> Because for the MQ-7 you need to:
>
>    Apply 5V for 60 seconds, don't use these readings for CO measurement.
>    Apply 1.4V for 90 seconds, use these readings for CO measurement.

Ok, in that case it gets a bit more complicated, however still possible.

First however you need to figure out the power supply. ESP can't apply 5V nor 1.4V on any pin - that needs external circuitry.

After that the steps would be like this:
- Set a `never` update interval on the ADC sensor (we will trigger updates manually).
- Create an `interval` component, that:

  1. Enables power (switch)
  2. Waits a bit (delay)
  3. Triggers ADC update (component.update)
  4. Waits another bit
  5. Disables PSU

kgalilio, 2019-04-16:
> Ok, in that case it gets a bit more complicated, however still possible.
>
> First however you need to figure out the power supply. ESP can't apply 5V nor 1.4V on any pin - that needs external [...]

I use NodeMcu V3 Lua with NodeMcu base. Using an external 5V power supply, I have abundance of pins to tap from the 5V source in addition to the 3.3V.
The problem is cycling the power. Even with a voltage regulator, it would require a dedicated circuit logic to cycle the power between the 5V and 1.4V.

OttoWinter, 2019-04-16:
> The problem is cycling the power. Even with a voltage regulator, it would require a dedicated circuit logic to cycle the power between the 5V and 1.4V.

Exactly, you need external circuitry. Once you've got the external circuitry you can look into how you would control it from software - if it's just simple binary GPIO signals it will be relatively simple.

kgalilio, 2019-05-24:
Managed to get something going, but I need help with Templating Actions.
As far as the hardware is concerned, I used pin D6 as an output GPIO which is connected to a 50k potentiometer. The regulated output of the potentiometer is connected to an [adjustable voltage regulator](https://www.onsemi.com/pub/Collateral/KA278RA05C-D.pdf).
When GPIO12 is low, the voltage is not regulated (I am using an external 5V power supply to power both the D1 mini pro and the MQ-7), hence the MQ-7 gets the 5V.
When GPIO12 is high, the 3.3V frpm GPIO12 going to the potentiometer creates a resistance which makes the adjustable voltage regulator drop to 1.4V.

As far as the switching cycle between 60s heating at 5V and 90s reading at 1.4V, I used the following code:
```
output:
  - platform: gpio
    pin: D6
    id: 'generic_out'
switch:
  - platform: output
    name: "Switch LDO2"
    id: sw_ldo2
    output: 'generic_out'

interval:
  - interval: 150s
    then:
      - logger.log: "The sensor is heating!"
      - delay: 1min
      #- template action to measure the sensor value should be here
      - switch.turn_on: sw_ldo2
      - logger.log: "The sensor is measuring!"
      - delay: 90s
      - switch.turn_off: sw_ldo2
```

Not the tidiest code I am sure, but I can visualize what is happening.
Now to get the sensor reading, I need to use a template action to read the MQ-7 Analog reading right at the end of the 1 minute heating cycle before switching to 1.4V.

The following code is good for continuous measurement. How can I make it measure once only at the end of the 1 minute heating cycle?

```
  - platform: adc
    pin: A0
    name: "Master Bedroom cO concentration"
    update_interval: 60s
    filters:
      - lambda: return (x * 1024/5); // WITHOUT PROPER CALIBRATION YOU WILL NOT GET SERIOUS RESULTS
    unit_of_measurement: "ppm"
    icon: 'mdi:fire'

```

OttoWinter, 2019-05-24:
@kgalilio See https://github.com/esphome/feature-requests/issues/159#issuecomment-483378729 - update_interval never and component.update

Also, see calibrate_linear for calibration (instead of lambda filter).