esphome/feature-requests #160: PCF8574 LCD backlight control.

hmmhhh, 2019-04-07:
**Describe the problem you have/What new integration you would like**

PCF8574 LCD display backlight control. ON/OFF and brightness.

**Please describe your use case for this integration and alternatives you've tried:**

Sonoff-tasmota has ON/OFF function for this display.

**Additional context**

Goal is to automate display backlight to save energy. Éxample use in sauna where temperature goes above 50C turn backlight ON.

rradar, 2019-04-07:
actually I did this with a hardware hack. There is this little jumper on the i2c controller board. I took one of them and wired to my esp8266 (pin D5). I also wired a pir sensor on D7 - when motion is detected the lcd background light is automatically turned on (and turns of after a delay)

This is my code:
```

output:
  - platform: gpio
    pin: D5
    id: lcdbacklight

switch:
  - platform: output
    name: "LCD LED"
    output: lcdbacklight
    id: switch_lcdbacklight

binary_sensor:
  - platform: gpio
    pin: D7
    name: "PIR"
    device_class: motion
    filters:
      - delayed_off: 3000ms
    on_press:
      then:
        - switch.turn_on: switch_lcdbacklight
    on_release:
      then:
        - switch.turn_off: switch_lcdbacklight
```

hmmhhh, 2019-04-07:
Do you use 5V or 3.3V on display? I think that if you use 5V and put that on 3.3V pin you fry the wemos or nodemcu etc.

rradar, 2019-04-07:
@hmmhhh I don't have fries and it runs since a long time. Actually I think the display is 5V. Wasn't the esp8266 5V 'tolerant' (confirmed by the CEO of Espressif himself)?

hmmhhh, 2019-04-08:
@rradar Still software solution would be nice.

rradar, 2019-04-08:
@hmmhhh sure why not. When it's possible to toggle it over the i2c bus this is for sure the most elegant solution!

danielhad, 2019-04-09:
Ohh yes, a software-solution would be very very nice! But meanwhile will try @rradar solution.

hmmhhh, 2019-05-18:
I got hardware hack to work. 

Here is simplified instructions to do it.

I have nodemcu.

Connect display to nodemcu:

Display VCC: --> nodemcu VIN (5V)
Display GND: --> nodemcu GND
Display SDA: --> nodemcu D1 (example)
Display SCL: --> nodemcu D2 (example)

Display LCD jumper negative pin  (use multimeter, mine was upper pin if text is right way) --> nodemcu D5 (example)

No connection to display positive pin (only one wire from nodemcu to display)

espconf configuration

i2c:
  sda: D1
  scl: D2
  scan: True

display:
  - platform: lcd_pcf8574
    dimensions: 20x4
    address: 0x27
    lambda: |-
      it.strftime(0, 0, "%d.%m.%Y %H:%M", id(sntp_time).now());
// displays current time on first line of display

output:
  - platform: gpio
    pin: D5
    id: lcdbacklight

switch:
  - platform: output
    name: "LCD LED"
    output: lcdbacklight
    id: switch_lcdbacklight

Still software solution would be nice. Example display dimming.

zeddy1, 2019-05-19:
I've been playing with this. With the config below (and backlight jumper in place), behaviour is that the backlight is on, but toggling this switch turns it off momentarily. The HA switch doesn't recognise the backlight turning back on.

Not a solution, but software control is clearly there somewhere.

display:
  - platform: lcd_pcf8574
    dimensions: 18x4
    address: 0x27

pcf8574:
  - id: 'pcf8574_hub'
    address: 0x27
    pcf8575: False

switch:
  - platform: gpio
    name: "PCF8574 Pin #7"
    pin:
      pcf8574: pcf8574_hub
      number: 7
      mode: OUTPUT
      inverted: true

adarazs, 2019-06-01:
As you can see from the mentions I implemented this on the `lcd_pcf8574` (with the help of @OttoWinter, thanks!) and using it to control the backlight based on motion with a timeout, so same as @rradar's use case.

For reference, this is how I solved it (assuming the display's id is `mydisplay`):

    binary_sensor:
      - platform: gpio
        pin: D5
        name: "ESP Motion"
        id: motion
        device_class: motion
        on_press:
          then:
            - lambda: |-
                id(backlight).publish_state(true);
                id(backlight).publish_state(false);
      - platform: template
        id: backlight
        filters:
          - delayed_off: 90s
        on_press:
          then:
            - lambda: |-
                id(mydisplay).backlight();
        on_release:
          then:
            - lambda: |-
                id(mydisplay).no_backlight();

This way I get a 90 second timeout after the last motion ON event. You can also use `it.backlight()` and `it.no_backlight()` from within the display lambda in case you want to control the backlight from there.

This is not merged yet, so you need to wait for this to trickle down into a release, or get my pull request. Let me know if you come up with a more neat solution for the display timeout than what I did.

closed 2019-06-03 by OttoWinter