esphome/feature-requests #163: Binary Sensor Component - logic if button is pressed.

tunip, 2019-04-08:
**Describe the problem you have/What new integration you would like**
I would use a ESP doorbell device (https://www.tindie.com/products/ErikLemcke/doorbell-modernizr/) with ESPHome.

If I use it with ESPHome (binary_sensor, on_press) I got 3-7 messages if the doorbell button is pressed.

With the sample code from the vendor I recevie only one message if the doorbell button is pushed.

The part of the original Arduino code is very simple at this point :
```
if ( doorbellState == LOW ) {
    client.publish(doorbell_topic, "on" , true);
    Serial.println("Doorbell is pressed!");
    delay( 5000 );
    client.publish(doorbell_topic, "off" , true);
}
```

I also tried a lot of combinations with the debounce values (delayed_on, delayed_off) in ESPHome.
But it is not the same logic as from the arduino code above.

**Please describe your use case for this integration and alternatives you've tried**
I am not able map the logic of the Arduino code into a binary_sensor in ESPHome. So would it be possible to add
a feature which maps the logic from the Arduino code like this?

1. Press the doorbell (no matter if the transformer bounces multiple times on button press)
2. change state ON
3. wait a specific time (no matter if i press the in this time multiple times)
4. change state OFF

Maybe it could be done with "lambda".

brandond, 2019-04-08:
Try https://esphome.io/components/binary_sensor/index.html#binary-sensor-on-click instead of on_press, it has debounce logic built in.

tunip, 2019-04-08:
Will try that later...

```
binary_sensor:
  - platform: gpio
    pin:
      number: GPIO16
      mode: INPUT_PULLUP
    on_click:
      min_length: 10ms
      max_length: 5000ms
```

OttoWinter, 2019-04-09:
> I also tried a lot of combinations to debounce with delay, delayed_on, delayed_off, etc. in ESPHome.

Could you include which ones you tried? I think delayed_off: 5s would get quite close to the behavior of the original device. Also, I would not really call the original behavior "debouncing". It's more like a really slow poll interval.

tunip, 2019-04-11:
Many thanks, that was the solution.

closed 2019-04-11 by tunip