esphome/feature-requests #164: Serial port GPS Library (tinygps?)

tubalainen, 2019-04-08:
**Describe the problem you have/What new integration you would like**

tinygps library or similar (GPS serial port lib)

**Please describe your use case for this integration and alternatives you've tried:**

The possibility to create a wifi based GPS tracker that could report location in accordance with the device_tracker component in HASS. I would personally use this to track my dumb robotic lawn mower. Output could be MQTT in accordance with device_tracker json_mqtt component in hass or similar.

**Additional context**

tubalainen, 2019-04-15:
Or the owntracks component hence we can use more attributes :)

OttoWinter, 2019-05-27:
GPS for time source has been added in 1.13 - location info should not be too hard to add since the base exists now.

closed 2019-06-06 by OttoWinter