esphome/feature-requests #165: Support hass.io ingress

martijnrondeel, 2019-04-08:
**Describe the problem you have/What new integration you would like**
Please look into implementing hass.io ingress support, like they did for configurator (https://github.com/home-assistant/hassio-addons/pull/560).

**Please describe your use case for this integration and alternatives you've tried:**
It would allow ESPHome to run on the same port as HASS and removes the need for reverse proxy setups.

OttoWinter, 2019-04-09:
For that:
- ingress needs to be documented - I don't want to try to reverse engineer it (also if it's not documented there might be sudden breaking changes in the format)
- Host networking mode could complicate things
- Websocket support in ingress (might already exist, but again not documented)
- Authentication would need to change. Under the ingress port authentication can be disabled while it needs to be enabled on the "default" one.

dattas, 2019-04-15:
Looks like they've officially released/documented ingress and I would also like to see this implemented. https://www.home-assistant.io/blog/2019/04/15/hassio-ingress/ for the annoucement and https://developers.home-assistant.io/docs/en/next/hassio_addon_presentation.html#ingress for the short implementation details. It does say it supports websockets. Sounds like the only remaining hurdle is ensuring the devices can still communicate with esphome. I'd imagine this could be implemented by keeping the existing setup and simply adding the extra port to listen on for the ingress implementation (and setting that port to only accept from the right ip + assume authenticated if so).

closed 2019-04-24 by OttoWinter