esphome/feature-requests #166: Support for Sim800 module

adonno, 2019-04-09:
**Describe the problem you have/What new integration you would like**

Is there a way to add support for the Sim800 Modules ?  

**Please describe your use case for this integration and alternatives you've tried:**

I want to build an esphome based alarm and would like to have a backup in case network access fails. Having a Sim800 module would enable me to send SMS or Trigger Calls in case of Alarm 

**Additional context**

OttoWinter, 2019-04-09:
Just FYI, this will probably not happen any time soon (no abstraction for notify components exists). Though it might not be too hard with custom components.

glmnet, 2019-04-12:
I might work on something like this in the near future. @OttoWinter is there a preferred way of sharing custom components other than the cookbook documentation?

OttoWinter, 2019-04-15:
> I might work on something like this in the near future. @OttoWinter is there a preferred way of sharing custom components other than the cookbook documentation?

I would prefer a standalone GH repo, because: that is more flexible (if you fix/change something, you won't need to update docs repo).

glmnet, 2019-04-23:
[image]
This is the SIM800L Module, I have one and did some custom code for it to run on Arduino Pro Mini.
The module can 
* Make / Receive calls,
* Send / Receive SMS text messages (~160 chars otherwise coder have to join them)
* GPRS internet communication (I'm not sure of data rate but it's not fast)

I would like to keep using it as a SMS Send / Receiver.

How could this integrate in ESPHome?
I don't know if text_sensor has some fit for it, or should I go completely with a new custom component? Which would be the services it could implement? And the notifications?
E.g. one real case use I gave it was to arm / disarm an alarm system, that would work by the module receiving a custom SMS message from a specific phone sender (my phone)

It might have a templatable action for SMS received and a simple _SmsCall_`.Perform()` which should collect receiver number and text message.

````yaml
sms_sender:
  - platform: SIM800L
    id: sms
    on_sms_received: 
      - lambda: |- ....

binary_sensor:
  - platform: ...
    on_press:
      - lambda: |-
        auto call = id(sms).receiver("+549...");
        call.text = "Hello there";
        call.perform();

OttoWinter, 2019-04-24:
> I don't know if text_sensor has some fit for it, or should I go completely with a new custom component? 

Nah, text_sensor doesn't make much sense here - text sensors are for long-lived states, but this thing is works through events. I'd say in this case we probably don't need to create a base abstraction for this component and doing it directly via automations is fine.

> Which would be the services it could implement? And the notifications?

At first, I think we should concentrate on SMS, since those are the simplest.

So:
- A trigger, like `on_sms_received` where the user can do whatever they want (like publish events, toggle a template binary sensor etc)
- An action, like `sim800l.send_sms`, that has the receiving phone number and text included.

I don't know how phone calls/gprs would be done architecturally, so I don't think it should be a focus for now.

glmnet, 2019-08-04:
I’m closing this as the sim800l is in dev branch.

closed 2019-08-04 by glmnet