esphome/feature-requests #168: ESP32 Camera on screen data / overlay (i.e. date/time)

daenny, 2019-04-09:
**Describe the problem you have/What new integration you would like**
I would like to overlay the current time/date overlaid on the camera image. 
I have an automation which takes a snapshot and sends me the picture via telegram, but I had it sometimes that the picture was out-dated. So a capture time overlaid on the image (as for instance zoneminder does, see below) would be nice to see that the picture is updated. 

[image]

**Please describe your use case for this integration and alternatives you've tried:**

**Additional context**

nickrout, 2019-04-09:
I winder if this is something best left to home assistant?

OttoWinter, 2019-04-09:
Not possible. The ESP just forwards the already-encoded JPEG directly from the camera module to HA. Because it's already encoded, we can't change any pixels directly - that would only be possible if the camera module could do that by itself.

closed 2019-04-09 by OttoWinter

daenny, 2019-04-09:
I have seen some examples which paint rectangles on detected faces. But they then probably decode and encode the image, I guess, and I see why this might be a bit out-of-scope.  
Ok, maybe it would possible to add that "on receive" in the esphome/camera component in HA?