esphome/feature-requests #172: pn532 on_no_tag

AndreasRavn, 2019-04-11:
**Describe the problem you have/What new integration you would like**
I would like to propose a on_no_tag, triggering when no tags are near

**Please describe your use case for this integration and alternatives you've tried:**
I'm using the pn532 with the on_tag to update a template text_sensor. Detects fine, but I can't use it to detect when the card leave the proximity of the sensor. The on_tag only gets called when a card is near. At the moment I just blank out the text_sensor at regular intervals, but that gives me multiple triggers if the card is still near.

**Additional context**
I think the binary sensors already does kind of this, but I don't wan't to set up binary sensors for 20+ cards.

OttoWinter, 2019-04-15:
Well the thing is that the core itself also uses a timeout for detecting when no tag is present.

You can use a script that you stop/restart when on_tag gets called that clears the text sensor after a while (see scan_interval setting).

AndreasRavn, 2019-04-17:
Cool, thanks, its working for me. Still think the on_no_tag would make sense, but as long as there is a solution.

closed 2021-05-17 by jesserockz