esphome/feature-requests #173: Support water flow sensor

postlund, 2019-04-11:
**Describe the problem you have/What new integration you would like**
I'm building a an automatic watering system for my yard and I would like to monitor if water is flowing or not, mostly to detect leaks but also to get some insights in how much water is used. My plan is to use esphome to control relays (connected to the valves) but also for measuring the water flow using a water flow sensor.

**Please describe your use case for this integration and alternatives you've tried:**
I have not found any sensor that does this.

**Additional context**
I'm using this sensor in particular (swedish):

https://www.electrokit.com/produkt/flodesmatare-1-30l-m-1-2/

But I guess most sensors work in the same way (outputs pulses). A simple Arduino sketch is here:

https://www.instructables.com/id/How-to-Use-Water-Flow-Sensor-Arduino-Tutorial/

Might exist arduino libraries that are part of platform.io as well.

glmnet, 2019-04-12:
Sounds like you can use the [pulse counter](https://esphome.io/components/sensor/pulse_counter.html) component as this device does not require any special electronic interface. May I been missing something otherwise?

postlund, 2019-04-14:
Ah, right. That might actually work. Would just have to figure out a good filter to match my valves. I'll see what I can do and report back, thanks!

OttoWinter, 2019-04-15:
Yes, the pulse counter covers this use case, see energy meter cookbook for and example filter. (when done, a cookbook entry for water flow meters would also be nice).

closed 2019-04-15 by OttoWinter