esphome/feature-requests #175: Support for 802.11 LR (Long Range Wifi for distances >1km)

rradar, 2019-04-12:
**Describe the problem you have/What new integration you would like**
I want to bridge a distance which can't be satisfied with a classical wifi (ap) coverage because of the lack of infrastructure (energy for example).

**Please describe your use case for this integration and alternatives you've tried:**
There is some context here:

> The 802.11 LR mode is a patented custom mode that can achieve a 1 km line of sight range so long as both the station and the soft-AP are connected to an ESP32 device. While these sort of ranges, and far more can be done with directional antennas and other hackery, this is an amazing accomplishment for an out-of-the-box board.

> While the LR mode is in the IDF documentation, there’s been only sporadic discussion of it in the forums since it was added to the frameworks. Although some people managed to pick up on the inclusion and have been testing the long-range mode in the field.

> Of course, what you’re sacrificing for additional range is data rate, the proprietary 802.11 LR mode designed for low-data rates—I’m guessing that the LR actually stands for “Low Rate” rather than “Long Range”—and I’ve yet to see much hard data about performance.

Source: https://blog.hackster.io/long-range-wifi-for-the-esp32-9429ab89f450

**Additional context**
Not yet

OttoWinter, 2019-04-15:
As the article says, this long range mode is only for ESP-to-ESP communication - ESPHome does not support such a communication model (see also FRs about using ESP as a wifi repeater).

It's a nice feature for sure, but it doesn't fit into ESPHome (plus I think the 1km is marketing talk, if someone really could get 1km out of wifi why don't more devices have it)

closed 2019-04-15 by OttoWinter

rradar, 2019-05-09:
About marketing: It really looks there is more than 1km of (line of sight) possible... :open_mouth: 

https://blog.hackster.io/long-range-wifi-for-the-esp32-9429ab89f450