esphome/feature-requests #177: support for passiv buzzer

Melcos1970, 2019-04-13:
Support for this shield with a buzzer (and other buzzer)
https://wiki.wemos.cc/products:d1_mini_shields:buzzer_shield
**Describe the problem you have/What new integration you would like**

get the buzzer to play a tone with a freq ex. 1000Hz or a note C6 (1047Hz)
and play a seriel of notes or freq. 
int freq[]={1047,1175,1319,1397,1568,1760,1976,2093};//Note name: C6 D6 E6 F6 G6 A6 B6 C7   http://newt.phys.unsw.edu.au/jw/notes.html
String note[]={"C6", "D6", "E6", "F6", "G6", "A6", "B6", "C7"};

play(note,time)
play(freq,time)
play("c6",1s,"d6",50ms)
play(1047,1s,1175,50ms)

in somekind of yaml

I don't know how or If it is posibel.
**Please describe your use case for this integration and alternatives you've tried:**

**Additional context**

https://wiki.wemos.cc/products:d1_mini_shields:buzzer_shield
can play Frequency: 1kHz-3kHz and use 1 pin. D5 to D8
the link contains

Buzzer Shield example
Schematic V1.0.0
MLT-8540 Datasheet

OttoWinter, 2019-04-15:
See [`set_frequency`](https://esphome.io/api/classoutput_1_1_e_s_p8266_p_w_m_output.html#a1744b85aa653e3155874d7253a471afd) method on ESP8266 PWM output (trigger from a lambda with `id(my_output).set_frequency(1047);`)

closed 2019-04-15 by OttoWinter

Landrash, 2019-04-18:
Even tough this is a solution @OttoWinter support for Buzzers/speakers without having to resort to lambda would open up a whole new set of devices :)

OttoWinter, 2019-04-22:
@Landrash Yeah, I guess I will add a `output.esp8266_pwm.set_frequency` action for that.