esphome/feature-requests #178: Omron Electronics D6T Series MEMS Thermal Sensors

coolbombom, 2019-04-14:
**Describe the problem you have/What new integration you would like**
I want my lights to turn on/off automatically in my restroom and as u can imagine a PIR sensor is no good. Therefore i would love to see a integration with Omron Electronics D6T Series MEMS Thermal Sensors.

**Please describe your use case for this integration and alternatives you've tried:**
I'm using a PIR sensor now but it is not working optimal.

**Additional context**
Arduino example code:
http://forum.arduino.cc/index.php?topic=217394.0
https://github.com/jordanlui/Omron-Arduino

kind regards
c_bb

mKeRix, 2019-05-18:
I made a custom component for this that works quite well for me, in my tests I could detect people up to 5m away. I run a Sobel Filter over the data to detect gradients in the environment to account for overall temperature changes. Needs a 4x4 sensor though, I tested it with the D6T-44L-06 on an ESP32.

I put the code and config into a GitHub gist for now, although I plan to contribute an official component: https://gist.github.com/mKeRix/4fc553574af0a2d8682734695160b859

You can play around with the `THRESHOLD` value to get it just right for your setup. The current one is quite low to get max distance.

loongyh, 2021-01-07:
@mKeRix Do you intend to make it an official ESPHome integration?

mKeRix, 2021-01-07:
Initially I wanted to, but I ended up integrating it into [room-assistant](https://room-assistant.io) for my use case instead. Since that project is keeping me quite busy I don't see myself adding this to ESPHome officially, at least not in the foreseeable future. If somebody else plans to look at this I would recommend to adapt the implementation a bit (check the aforementioned project for inspiration) to get better results.

loongyh, 2021-01-08:
Does the [mean + deltaThreshold](https://github.com/mKeRix/room-assistant/blob/2a4a350a9f4e45ae31f3932c46ad97a605b809f1/src/integrations/thermopile/thermopile-occupancy.service.ts#L56) instead of this Sobel filter work better for you in room-assistant?

mKeRix, 2021-01-10:
For me it does. On the D6T it did not make as much of a difference in the results from my tests, but the GridEYE sensor greatly benefitted from the mean + deltaThreshold integration. Either way, Sobel results are a bit hard to interpret with these low resolution sensors imo. Both approaches work off the same base assumption, they just take different ways to get to the result.

loongyh, 2021-01-10:
Hmm so in a way, for such low resolution sensors, the Sobel filter is kinda overkill? The simple mean+deltaThreshold will achieve similar results with less complexity.

mKeRix, 2021-01-10:
Pretty much, yeah, at least for the D6T. Other thermopile sensors will see better results on mean+deltaThreshold, at least from what I tested.

loongyh, 2021-01-12:
Thanks! Will evaluate both your algorithms to come up with the ESPHome integration. Will you like to be added as a codeowner (such that you'll be listed as a contributor for this integration and also get notifications on any issues opened against it)?

mKeRix, 2021-01-14:
Feel free to include me if you want, but you don't need to credit me explicitly. I'm happy to answer issues regarding the sensor and the algorithm to the best of my knowledge though, so I guess for that it would make sense. 🙂

shubness, 2021-07-07:
any updates on this thread? Having issues with the ESPHome custom component - it does not move from the occupied state. how can I improve this?

loongyh, 2021-07-08:
Have you implemented mKeRix's [mean+deltaThreshold](https://github.com/mKeRix/room-assistant/blob/2a4a350a9f4e45ae31f3932c46ad97a605b809f1/src/integrations/thermopile/thermopile-occupancy.service.ts#L56) algorithm currently being used for Room Assistant?

I still have not found the time to get started working on this. Will be great if you can take over and get this project going.

bjeanes, 2022-04-28:
Also interested in thermopile usage in ESPHome. A quick glance at [the example code](https://github.com/omron-devhub/d6t-2jcieev01-arduino/blob/master/examples/d6t-32l/d6t-32l.ino) linked from the D6T datasheet seems to indicate it's just i2c, so perhaps a custom component isn't necessary.

On the other hand, something like the [MRT311](https://www.aliexpress.com/item/1005003539743345.html) is not i2c, but seems like [the MRT311 datasheet](https://www.winsen-sensor.com/d/files/MRT311.pdf) implies it could be used like any [NTC sensor](https://esphome.io/components/sensor/ntc.html). 

However, I don't see any indication that the MRT311 gives a "grid" of pixels so is not really comparable to the D6T series I think.

There is also the [MLX90640](https://www.melexis.com/en/product/MLX90640/Far-Infrared-Thermal-Sensor-Array) which is more like the D6T and is also I2C, so should be simple to integrate.

The M6T is 4.5-5.5 VDC though and I don't have I2C experience really so I don't know if means the I2C voltage will need to be divided to avoid frying the ESP.

closed 2024-03-26 by coolbombom