esphome/feature-requests #179: Support for FC-28 soil moisture sensor

akajester, 2019-04-15:
**Describe the problem you have/What new integration you would like**

For these inexpensive soil moisture sensors to be a supported device in esphome

**Please describe your use case for this integration and alternatives you've tried:**

I'd like to monitor soil moisture and utilize esp8266 and home assistant to gather data and send alerts, etc. I see no other hardware to do this unless using xiaomi or some other platform. These sensors are inexpensive and would be a great addition to esphome.

**Additional context**

I found sample code for this sensor module here;

OttoWinter, 2019-04-15:
This is an analog sensor, and thus already supported with adc platform. See TEMT6000 example for converting the voltage values to moisture levels.

closed 2019-04-15 by OttoWinter

schmidiij, 2019-05-03:
Do you have a working config for it?

marrobHD, 2019-05-08:
Anyone have it got working?

Tim-Dixon, 2019-05-18:
I've got the sensor displaying values in voltage but I haven't figured out how to relate that to "Moisture Level" yet. It seems that I need to multiply the data to get it to a value of 100 between 0 and 3 volts as that is the 0 and 100 in relation to capacitance? Correct me if I'm wrong :o

OttoWinter, 2019-05-27:
@Tim-Dixon See calibrate_linear.

mikenabhan, 2019-06-06:
Here is the config that I use.  You will still have to calibrate to your specific sensor though.  What you need to do is get the raw voltage without any filters and then mark down the voltage for High and Low moisture.  Then add those into your lambda.  You will have to do this for each sensor as even the same batch varies slightly.  

`  - platform: adc
    unit_of_measurement: "%"
    icon: "mdi:flower-outline"
    accuracy_decimals: 0
    pin: A0
    filters:
      - lambda: |-
          if (x > 0.7) {
            return 0;
          } else if (x < 0.38) {
            return 100;
          } else {
            return (0.7-x) / (0.7-0.38) * 100.0;
          }
    name: "Soil Moisture Level"
    update_interval: 300s`

OttoWinter, 2019-06-07:
@mikenabhan Or.. this would be the same (minus the bounds check):

```yaml
  - platform: adc
    unit_of_measurement: "%"
    icon: "mdi:flower-outline"
    accuracy_decimals: 0
    pin: A0
    filters:
    - calibrate_linear:
	    - 0.38 -> 100
		- 0.7 -> 0
	name: "Soil Moisture Level"
    update_interval: 300s
```