esphome/feature-requests #180: Support for CCS881

dougmoscrop, 2019-04-15:
**Describe the problem you have/What new integration you would like**

I would like to see support for the [CCS811](https://cdn.sparkfun.com/assets/learn_tutorials/1/4/3/CCS811_Datasheet-DS000459.pdf) sensor.

**Please describe your use case for this integration and alternatives you've tried:**

I am automating a small grow-op and would like to monitor CO2 levels, mostly for curiosity. I've already purchased the sensor and tested it with PlatformIO, but integration with HA this way seems like it would be way better than rolling my own.

**Additional context**

This sensor needs to "warm up" for 20 minutes before reading values, I don't know if the concept of a sensor not being ready for an extended period of time is a new concept to the domain?

There are two libraries, I've tested with the simple Adafruit one and it worked, but the Sparkfun one looks like it has been updated more recently.

OttoWinter, 2019-04-15:
Duplicate of https://github.com/esphome/feature-requests/issues/19

closed 2019-04-15 by OttoWinter

dougmoscrop, 2019-04-15:
D'oh, I searched for 881, then fixed my typo (as you see) after the fact. Sorry.