esphome/feature-requests #181: ESP Soil Moisture Sensor

madrose, 2019-04-15:
**Describe the problem you have/What new integration you would like**

Is it possible to add a support for the following sensor:
https://www.tindie.com/products/aprbrother/esp-soil-moisture-sensor-rev-21/
This sensor uses TMP112 with I2C which is not support by default
**Please describe your use case for this integration and alternatives you've tried:**

I have tried to create a custom sensor but that turned out to be more complex than it seems for my beginner C++ programming skills. 

I have also checked 
https://github.com/esphome/feature-requests/issues/179
but such an emplementation will be limited to the manual calculation for the moisture but will miss the temp sensor (tmp112)

Being a fan of ESPHome and I have implemented it through out my home it only makes sense to have a proper implementation for such a device.

thanks in advance.

**Additional context**

[image]

OttoWinter, 2019-04-15:
The soil moisture calculation will already work as per #179.

So this feature request is about the TMP112 specifically.

madrose, 2019-04-15:
I guess that should be the case. tmp112 is not implemented.

marrobHD, 2019-05-08:
But what is the template

TheEggi, 2019-06-27:
What is the use case of that one over the xiaomi plant sensor?