esphome/feature-requests #184: Remove 23-character MQTT client_id limitation

JSteeleIR, 2019-04-18:
**Describe the problem you have/What new integration you would like**

ESPHome should allow custom MQTT client_ids that exceed 23 characters.

Cloud providers (GCP, AWS) offer services implementing MQTT brokers, allowing for IoT deployments with no local component. 

For these providers, the client_id is used to tie an individual device back to the cloud project/account, and as such, it can be relatively lengthy. In[ GCP for example, the client_id must be set to ](https://cloud.google.com/iot/docs/how-tos/mqtt-bridge#iot-core-mqtt-publish-python) "projects/{project-id}/locations/{cloud-region}/registries/{registry-id}/devices/{device-id}" which, before variable substitution is 41 characters. This obviously exceeds the limitation, meaning that the cloud IoT platform can't be used.

**Please describe your use case for this integration and alternatives you've tried:**

Extending/eliminating this restriction would enable ESPHome to be used in "remote" devices, with no local broker, instead leveraging a cloud-backed mqtt broker as a relay/master for more complex applications. 

For my specific case, I want to be able to have a remote ESPHome that communicates with the GCP MQTT broker, which I can publish to from a local HomeAssistant instance, or any arbitrary application that can integrate with the Cloud platform. This way, there's no need to host/expose an additional MQTT broker, and the security assertions that a provider-backed broker provides can be gained for free.

**Additional context**

Digging through the ESPHome component code and the backing AsyncMqttClient library, I can't seem to find a technical reason for this limitation.
The AsyncMqttClient library [lists no length requirements](https://sourcegraph.com/github.com/marvinroger/async-mqtt-client@master/-/blob/src/AsyncMqttClient.hpp#L46:20) regarding the client_id, and the ESPHome cpp component for the mqtt_client itsel[f doesn't seem to implement this limitation](https://sourcegraph.com/github.com/esphome/ESPHome@6682c43dfaeb1c006943ae546145e5f22262cadb/-/blob/esphome/components/mqtt/mqtt_client.h#L213:73&tab=references). In fact, aside from a [comment in the header](https://sourcegraph.com/github.com/esphome/ESPHome@6682c43dfaeb1c006943ae546145e5f22262cadb/-/blob/esphome/components/mqtt/mqtt_client.h#L44:3) and[ within the python validation check](https://sourcegraph.com/github.com/esphome/ESPHome@6682c43/-/blob/esphome/components/mqtt/__init__.py#L105:93) there doesn't seem to be any other reference/technical limitation I can find disallowing this.

OttoWinter, 2019-04-21:
The reason for this was that the [MQTT 3.1.1 spec](http://docs.oasis-open.org/mqtt/mqtt/v3.1.1/csprd02/mqtt-v3.1.1-csprd02.html) says that

> The Server MUST allow ClientIds which are between 1 and 23 UTF-8 encoded bytes in length, and that contain only the characters
> 
> "0123456789abcdefghijklmnopqrstuvwxyzABCDEFGHIJKLMNOPQRSTUVWXYZ" [MQTT-3.1.3-5].

So MQTT brokers are not *required* to support anything longer than 23 bytes. I guess that limitation can be removed though as most brokers (mosquitto in particular) seem to support it.

JSteeleIR, 2019-04-22:
Ah. That explains that. 

Given that the underlying code (aside from the validation layer) doesn't enforce this, what would be your preference on removing the limitation? Extend the validation to a higher limit? Remove the length check entirely? Warn on IDs longer than 23 but proceed anyway? 

I'm happy to send the PR if you have a particular preference. I lean towards removing the limit or warning (but I'm not sure how to implement the warning best).

OttoWinter, 2019-04-22:
> Given that the underlying code (aside from the validation layer) doesn't enforce this, what would be your preference on removing the limitation? Extend the validation to a higher limit? Remove the length check entirely? Warn on IDs longer than 23 but proceed anyway?

Just remove it, already done as part of https://github.com/esphome/esphome/pull/491 though

closed 2019-04-22 by OttoWinter