esphome/feature-requests #185: Dimming switch/rotary encoder

erazor666, 2019-04-21:
**Describe the problem you have/What new integration you would like**

I have quite a lot of LED strips based on ESPHome, and they work well and good. I control scenes and brightness with Home Assistant or binary sensors. I've yet to come across a reliable dimming solution based on holding/pressing buttons or a rotary encoder.
Is there any way to make a function/component in ESPHome that deals with this? 

This is much a ease of use thing and a big WAF (wife acceptance factor) for me at least. 
(I realize this may be possible with the current rotary encoder implementation, I have mostly regular impulse light switches in the house and they work great for binary sensors, just wish they were smarter. Its somewhat troublesome to mod wall boxes and get a hold or pretty looking caps for rotary encoders).
I guess this also somewhat related to the API server in the horizon. If to control dimming by holding the button pressed. I have not been able to get a good robust solution with Home Assistant automations either.
If anyone has code examples I would love to test it also.

Tl;dr
Dimming solution for LED lights driven by ESPHome by holding a button down
and/or
Start a dim "cycle" that you can stop at the desired brightness ie. click a button once to start the cycle, click once more to stop the cycle. The light would increase by x % per y seconds or something in that manner
 

**Please describe your use case for this integration and alternatives you've tried:**

**Additional context**

RGN01, 2019-04-21:
This is something I've tried and failed to do as well - although slightly different to mirror commercial dimmers that work like this.

- short press = toggle light on / off
- long press = brighten while pressed
If button is released and then another long press starts in 1 second then the behaviour changes to dim the light. If the button is again released and pressed within 1 second again it goes back to brightening the light. Release button to hold state.

Suggest rate of dimmer / brightening could be a variable.

And , ideally, it should be able to do this for lights attached to HASS.IO, rather than to lights attached to the ESP.

I hope that is similar enough to the action you need, @erazor666 - if not, I apologise and will create a separate feature request.

erazor666, 2019-04-22:
First of all I'm looking for a solution that works, and is reliable.
Actually that is a good thing if it could control any light that is attached to HA.
I didn't even think of it, that will make it much more useable.

How the action is executed is secondary to it actually exsisting at this point. To my knowledge nothing of this nature exsist today. I've tried on the HA side of things without results.

OttoWinter, 2019-04-22:
Well it's definitely possible to do this with automations.

```yaml
# in the button binary sensor
id: button
on_press:
  while:
    condition:
      binary_sensor.is_on: button
    then:
    - light.turn_on:
        id: id_of_the_light
        brightness: |-
          return id(id_of_the_light).get_current_values().brightness - 0.05;
    - delay: 0.05s
```

@pauly7300 has done this before, maybe he can share his config.

Anyway, I don't see a good way to simplify this. As the comments by @RGN01 show, there are so many different ways this "light dimmer control" could work. Plus then there's the question of how this would translate to YAML in a generalized way.

If someone has ideas for that, please do post them, until then this will remain a task that needs to be done in automations.

RGN01, 2019-04-22:
Thank you for your reply Otto. 

I will try to do this in automations - if I get it right I will share and will consider ways to translate to yaml - if I have any ideas I'll reply here. I'm a beginner so this could take a while!

pauly7300, 2019-04-22:
Yes, here's how i'm doing it.  
https://gist.github.com/pauly7300/70cc5291bbb0b3b3891f72943321d46d

It's not perfect but may give you some ideas.  The functionality is this.

When the button is pressed and held down, it changes fades up or down at a specific increment and length of time while the button is held down.  The next time the button is pressed, it fades in the other direction.  

I used a global variable to track the direction of the fade and flip the direction after releasing the button each time. 

The behavior works in the fade up direction fine (stopping at 100% even if I keep the button held) however in the fade down direction, when the brightness gets to zero, it has issues since i'm using the light.turn_on command to effectively turn off the light by setting the brightness to 0.  

@OttoWinter You thought it might make sense to add functionality through a .control action.  We discussed on 4/4/19 in the ESPHome discord server if any of you want to reference the conversation.

As for the LED Dimmer, I use Quinled-Quad boards.  https://quinled.info/2018/09/07/quinled-quad/
They work great and integrate very nicely with ESPHome.  They can control multiple strips at the same time with various voltages.

OttoWinter, 2019-04-24:
@pauly7300 Yeah the `light.control` has already been added in dev [here](https://github.com/esphome/esphome/blob/c7609ba5e7b7c5cc2216138bbc504e7a23998511/esphome/components/light/__init__.py#L387)

RGN01, 2019-04-24:
@OttoWinter a question on the code you posted above, if I may. It appears to be set for the light on the ESP, not on HomeAssistant so I've amended it as follows in an effort to make it work on HA. Unfortunately it is not working. I know the branch is working because I see the log entry and if I set a static brightness command the light changes accordingly - it is the variable aspect that isn't working. What am I doing wrong, please?

```
    on_press:
      while:
        condition:
          binary_sensor.is_on: Up
        then:
        - homeassistant.service:
            service: light.turn_on
            data:
              entity_id: light.study1
              brightness: |-
                return id(light.study1).get_current_values().brightness - 0.05;
        - logger.log: "Up pressed"
        - delay: 0.05s
```
I really appreciate any help. Thanks.

erazor666, 2019-04-24:
I guess when thinking about it, the way I tried to attack this with HA automations and using a binary sensor from a node, its taxing on the network, if the binary sensor is internal on the node it will not spam as much.
This again makes you dependant on the switch and the light being on the same node and thats not most use cases. 
I guess I touched on this when I made a feature request for the automation across nodes.
I read the design philosopy (the upcomming API server) on this and I agree on it for the most part.

This dimming function is a very desired function for many users and seeing as HA doesn't have anything really worthwhile, it would be very beneficial for many users. Though I'm not sure this is enough to re visit the issue, it would be great if a switch node could tell a light node to dim directly somehow.

I used to have a lot of Nexa 433mhz dimmers and switches. They worked like this when operated:
(2 buttons)

If the light was off, it would turn on from the on button

If the light was on it would start to brighten towards 100% then down to 0% and back an forth until you stopped it with another click of the on button. It would remember last setting when turned on/off

Last button would just turn it off. 

This was an ok solution and something i think could be replicated in code somehow. The off button could be a long press or something to have a one button solution.

I'm not sure on exactly how this can be archived, like a function in monochromatic lights on that side of things or if its something the switch node would control. My wild guess is that its best put on the node controling the light. So basicually you just ask the light to cycle up and down (interval adjustable perhaps) until stopped.

OttoWinter, 2019-04-24:
@RGN01 The `id()` wrapper only works for esphome-internal types. If you want to control a remote light, you need to use data_template and variable. However, it will probably be easier to write a script in HA for it. This thread here is for ESPHome-internal types only please.

RGN01, 2019-04-24:
Thanks, @OttoWinter  - and apologies for posting in the wrong area.

OttoWinter, 2019-04-25:
With the latest changes in esphome, this will be possible:

```yaml
on_press:
- while:
    condition:
    - binary_sensor.is_on: the_button
    then:
    - light.dim_relative:
        id: the_light
        relative_brightness: 1%
        transition_length: 0.01s
    - delay: 0.01s
```

which would already improve this a lot.

OttoWinter, 2019-05-28:
Ok, closing this issue as it has gotten very easy already with the new `dim_relative`.

closed 2019-05-28 by OttoWinter

Joshfindit, 2019-05-28:
@OttoWinter Is there a corresponding ‘brighten_realative’? Or does ‘dim_realative’ start brightening once it gets to zero?

OttoWinter, 2019-05-28:
> @OttoWinter Is there a corresponding ‘brighten_realative’? Or does ‘dim_realative’ start brightening once it gets to zero?

Please explain in more detail. dim_relative only dims by a given fixed amount. If that is positive the light will get brighter. If it's negative the light will get darker.

Joshfindit, 2019-05-28:
Aha, negative dimming fills the gap, thank you.