esphome/feature-requests #186: Integration with WebUSB

carldebilly, 2019-04-23:
**Describe the problem you have/What new integration you would like**
ESPHome Dashboard is very cool, but it would be a lot cooler if we would be able to provision USB devices directly from a web browser with the device connected to the client machine.

**Please describe your use case for this integration and alternatives you've tried:**

Nothing tested yet: just read the specs for now.

**Additional context**

It's now supported by Google Chrome and we can expect other browser to support it soon.
Doc: <https://wicg.github.io/webusb/>

OttoWinter, 2019-04-24:
Looked at this some time ago, but as far as I understand this requires specific firmware on the receiving device (at least in all the guides for similar products like arduino they all had to have webusb library in the fw).

If that is the case, then this is probably useless. Because when you manually need to flash the device with a webusb-able fw, why not flash it with esphome directly?

carldebilly, 2021-06-17:
I think the new _ESP Web Tools_ is filling the gap asked in this issue.

OttoWinter, 2021-06-17:
@carldebilly Correct (but it's using the new `WebSerial`, not `WebUSB`); if you install 1.19.0 it will already have the "~~flashing~~ installing over browser" in the dashboard.

closed 2021-06-17 by OttoWinter