esphome/feature-requests #187: Support custom mqtt state_on/state_on and payload_on/payload_

aqualx, 2019-04-23:
**Describe the problem you have/What new integration you would like**

Some systems use custom mqtt conventions. For example [homeA mqtt](https://github.com/binarybucks/homA/wiki/Conventions) convention use payload_on=1 (instead of "On") and payload_off=0(instead of "Off") 

**Please describe your use case for this integration and alternatives you've tried:**

In home-assistant it is possible to support such convention with next setup:
- platform: mqtt
  name:  Kitchen
  state_topic: /devices/thermostat/controls/kitchen_enabled
  command_topic: /devices/thermostat/controls/kitchen_enabled/on
  _payload_on: 1
  payload_off: 0
  state_on: 1
  state_off: 0_
  icon: mdi:radiator

**Additional context**

Can't find any possibility to support such mqtt convention natively in Esphome.

OttoWinter, 2019-04-24:
> Can't find any possibility to support such mqtt convention natively in Esphome.

If you're going to use this with Home Assistant, why use another convention? Doesn't entirely make sense for me.

Sure this can be implemented, someone just needs to do it.

aqualx, 2019-04-24:
My primary hardware controller uses homeA mqtt convention. Hass is only for user interaction...

davea, 2020-02-18:
I made a component (based on `MQTTSwitchComponent`) that tackles this exact problem: https://gist.github.com/davea/466d42d62e3ff32fd0e5dc16d7d525d0